osdp-dev/libosdp #2: Unit test produces segmentation fault

chayandashora, 2020-03-29:
Below is the terminal output:

`chayandashora@linux-ggds:~/Downloads/libosdp_master_sid/build> make && make check`
`[100%] Built target osdp`
`[100%] Built target osdpctl`
`[100%] Built target osdp`
`[100%] Built target osdp-test`
`------------------------------------------`
            `OSDP - Unit Tests`             
`------------------------------------------`
`Testing cp_build_packet(CMD_POLL) -- success!`
`Testing cp_build_packet(CMD_ID) -- success!`
`Testing phy_decode_packet(REPLY_ACK) -- success!`
`Testing cp_queue_command() -- success!`

`Starting CP fsm state tests`
    `-- executing test_cp_phy_fsm()`
`make[3]: *** [CMakeFiles/check.dir/build.make:57: CMakeFiles/check] Segmentation fault (core dumped)`
`make[2]: *** [CMakeFiles/Makefile2:68: CMakeFiles/check.dir/all] Error 2`
`make[1]: *** [CMakeFiles/Makefile2:75: CMakeFiles/check.dir/rule] Error 2`
`make: *** [Makefile:164: check] Error 2`

Upon further analysis, I found out that in test/test-cp-phy-fsm.c, line 126,
`while (result) {`
if the result value is 1, then segmentation fault appears.

cbsiddharth, 2020-03-29:
@chayandashora, thanks for reporting this issue. Can you provide some information on your dev environment? I'm unable to reproduce it in my Mac or Centos-7 VM. Besides, the CI is building this on ubuntu (trusty) without this issue.

chayandashora, 2020-03-29:
@cbsiddharth 
I am running OpenSuse Leap 15.1 on Oracle Virtual Box

cbsiddharth, 2020-03-29:
@chayandashora, I think this issue is now fixed. Please let me know if it persists.

commit 6d92a48a (2020-03-29): Fix #2 Unit test produces segmentation fault

A segfault was caused due to a hoisted copy of ctx pointer before it was
initilized by test_cp_phy_fsm_setup() in test/test-cp-phy-fsm.c
commit 0ea53d6a (2020-03-30): test: Fix another copy before assign bug (see Fix #2)
closed 2020-03-29 by cbsiddharth