osdp-dev/libosdp #7: Wrong SCS code sent by osdp_pd?

z8a, 2020-05-27:
I'm using OSDP.Net library to check your library and I had problems with Secure Channel.
Changing osdp_pd.c with:

diff --git a/src/osdp_pd.c b/src/osdp_pd.c
index 5e5b997..02c63d7 100644
--- a/src/osdp_pd.c
+++ b/src/osdp_pd.c
@@ -330,7 +330,7 @@ int pd_build_reply(struct osdp_pd *p, struct osdp_data *reply, uint8_t * pkt)
 
 	if (smb && (smb[1] > SCS_14) && isset_flag(p, PD_FLAG_SC_ACTIVE)) {
 		smb[0] = 2;
-		smb[1] = (len > 1) ? SCS_17 : SCS_15;
+		smb[1] = (len > 1) ? SCS_18 : SCS_16;
 	}
 
 	if (len == 0) {
it works.
Is it correct?

cbsiddharth, 2020-05-27:
Hi @z8a,  that is indeed a bug. Thanks for reporting it. Since you did the work of finding it, do you want to raise a PR for this?

If you don't want to, let me know and I will make this commit for you.

cbsiddharth, 2020-05-27:
A bit of background on how this bug managed to slip by all the testing (for my own reference). phy_decode_packet() was written to be reused between CP and PD so it was doing a more relaxed check at two places that allowed for this to happen.

Relaxed check on SCS_15 at osdp_phy.c line 310
```
	if (isset_flag(p, PD_FLAG_SC_ACTIVE) &&
		pkt->control & PKT_CONTROL_SCB &&
		pkt->data[1] >= SCS_15)
	{
```
Relaxed check on SCS_17 or SCS_18 at osdp_phy.c line 325
```
	if (pkt->data[1] == SCS_17 || pkt->data[1] == SCS_18) {
```

TODO: must fix these checks so we can catch faulty PDs.

commit 2f4f6607 (2020-05-27): Fix wrong SCS codes sent by PD implementation

phy_decode_packet() was written to be reused between CP and PD so it was
doing a more relaxed check at two places that allowed this bug to escape
uint and integration testing.

Fixes: #7

Reported-By: GitHub-User: z8a
closed 2020-05-27 by cbsiddharth