osdp-dev/libosdp #14: Add support for manufacturer specific commands (osdp_MFG/osdp_MFGREP) in CP and PD modes

cbsiddharth, 2020-09-03:


te-johan, 2020-09-03:
Thank you for this, nice additional changes as well.
A python callback to handle MFGREP would be good.
I’m not sure using a hard coded byte for size is a good idea. Does the receiver have to know the data size? I would assume that the size would be known by the implementation already.

cbsiddharth, 2020-09-03:
MFGREP can be handle from python, just need some additional work/thinking. Basically the callback handler must return a dict with keys populated for MFGREP. I can think of `return (ret_code, mfg_rep)` from the python command handler and then choose to send MFGREP based on return code but that looks ugly. So I deferred that part for now.

> I’m not sure using a hard coded byte for size is a good idea.

For now, LibOSDP now does make one enforcement to osdp_MFG: before the command data bytes, there must be a 1 byte length prefix. This is how the rest of the protocol is defined and IMO this is a miss from the spec side. Sending that one byte allows me to aggressively error check and do "informed" buffer copy. Do you see a problem with this design?

cbsiddharth, 2020-09-03:
On Second thought, there might be devices that have implemented osdp_MFG already and they won't understand this length prefix. Will fix it, thank you for bringing it up.

closed 2020-09-03 by cbsiddharth

cbsiddharth, 2020-09-05:
@te-johan Both MFG and MFGREP are now handled in pyosdp as well as LibOSDP. Can check if everything works well for your use case?

te-johan, 2020-09-07:
Thanks, I've tries out the PD part and it seems to work good.
Now when I update the CP end I get a crash when trying to send a MFG command from python.
the C code gets a bad pointer to the data object.

te-johan, 2020-09-07:
Not sure what the issue is. PyArg_Parse() in pyosdp_dict_get_bytes() sets data_bytes to NULL.
However the returned length is correct. I run python 3.8.5.

cbsiddharth, 2020-09-07:
Can you show the python dict you are trying to send? I'll try to reproduce this.

te-johan, 2020-09-07:
I actually just tried to send zero bytes.
        mfg_cmd = {
            "command": osdp.CMD_MFG,
            "vendor_code": 0x00000000,
            "mfg_command": 0,
            "data": bytes([ 00 ])
        }
I got it to work by using "y*" and read it into a Py_buffer instead. Documentation said it was the recommended way to work with data, but I don't know why it works with "y*" and not "y#"

cbsiddharth, 2020-09-07:
In pyosdp_dict_get_bytes(), length == 0 and buf == NULL are valid. So, with the below change, your mfg_cmd sends 1 byte of "0".

```diff
diff --git a/python/pyosdp_utils.c b/python/pyosdp_utils.c
index 259ab87..48198b7 100644
--- a/python/pyosdp_utils.c
+++ b/python/pyosdp_utils.c
@@ -148,10 +148,6 @@ int pyosdp_dict_get_bytes(PyObject *dict, const char *key, uint8_t **data, int *
        if (!PyArg_Parse(obj, "y#", &buf, &len))
                return -1;

-       if (buf == NULL || len == 0) {
-               PyErr_Format(PyExc_ValueError, "Unable to extact data bytes");
-               return -1;
-       }
        *data = buf;
        *length = len;
        return 0;
```

To send zero bytes of data use,  `"data": bytes()`

cbsiddharth, 2020-09-07:
Just to be sure, you are saying len was set to non-zero while  buf was NULL right? I'll read the docs again. ---- On Mon, 07 Sep 2020 19:22:10 +0530  notifications@github.com  wrote ----
Yes that or a similar change is indeed needed to be able to not send any data.
It however does not fix my issue. it still crashes because PyArg_Parse writes a NULL pointer into &buf.
This is the code that works for me.
index 259ab87..4329e6c 100644
--- a/python/pyosdp_utils.c
+++ b/python/pyosdp_utils.c
@@ -136,8 +136,7 @@ int pyosdp_dict_get_int(PyObject *dict, const char *key, int *res)
 int pyosdp_dict_get_bytes(PyObject *dict, const char *key, uint8_t **data, int *length)
 {
        PyObject *obj;
-       uint8_t *buf;
-       int len;
+        Py_buffer buffer;
 
        obj = PyDict_GetItemString(dict, key);
        if (obj == NULL) {
@@ -145,14 +144,14 @@ int pyosdp_dict_get_bytes(PyObject *dict, const char *key, uint8_t **data, int *
                return -1;
        }
 
-       if (!PyArg_Parse(obj, "y#", &buf, &len))
+       if (!PyArg_Parse(obj, "y*", &buffer))
                return -1;
 
-       if (buf == NULL || len == 0) {
+       if (buffer.buf == NULL || buffer.len == 0) {
                PyErr_Format(PyExc_ValueError, "Unable to extact data bytes");
                return -1;
        }
-       *data = buf;
-       *length = len;
+       *data = buffer.buf;
+       *length = buffer.len;
        return 0;
 }

te-johan, 2020-09-07:
Yes exactly, that is what I saw using GDB.

cbsiddharth, 2020-09-07:
Okay, I've given my fair bit of time to this and have no idea how this can happen (with just the documentation without locally reproducing it). But since that doc clearly mentions "y*" is the recommended way, and it works for you, can you please raise a PR with the change you have above? it looks good to me.

commit 818ae384 (2020-09-03): Add support for manufacturer specific commands

Fixes: #14
commit 4e29d705 (2020-09-03): Remove length prefix requirement from osdp_MFG

There might be devices that have implemented osdp_MFG already and they
won't understand the length prefix that LibOSDP enforces.

Fixes: #14