osdp-dev/libosdp #23: Info: what is necessary to save in pd_notify_keyset to allow secure connection after a power on?

z8a, 2020-09-14:
I think it is the data field of struct osdp_cmd_keyset but is it enough?
And at power on have I to pass it to osdp_pd_setup?

Thank you.
Paolo

cbsiddharth, 2020-09-14:
Yes when `type == 1`, the contents of `data` (16 bytes), must be stored in a persistent memory and passed to `osdp_pd_setup()` at boot as the SCBK parameter.

closed 2020-09-14 by cbsiddharth

z8a, 2020-09-17:
After I have connected my device with secure protocol if I restart control panel (an access control system) it will reconnect in secure mode but if I restart my device it will go to unsecure. In this condition I can see CMD_CHLNG message from control panel but not CMD_SCRYPT. Is there something that is not initialized after reset? Your library I'm using is not the last version but this is the only problem I have and, at the moment, is not so simple to upgrade it for my device.
Many thanks.

cbsiddharth, 2020-09-17:
From what you describe, I suspect the `struct osdp_pd_id` members are not initialized properly. When you call osdp_pd_setup at each boot on the PD side, the same `osdp_pd_info_t` must be passed. This along with the SCBK are used to setup the secure channel.

Alos, it's highly recommended that you use the latest version 1.3.0 which has lots of bug fixes and new feature additions. If you had to modify LibOSDP to fit your needs, please open issues for such problems you faced and describe how you solved it. This way I can fix them upstream  (or provide workarounds) so you can uptake newer versions easily.