osdp-dev/libosdp #24: Add support for "File Transfer Messages" for DFU over OSDP

cbsiddharth, 2020-09-18:


te-johan, 2020-09-21:
I find the standard rather incomplete in regards how file transfer messages are handled.
What command is used to send the file transfer? I have the SIA OSDP 2.1.7 and I've not seen it documented.

cbsiddharth, 2020-09-21:
Yes. This is a nice to have featured requested by another user by email. This is a tracking issue for that request. I've asked him this exact question along with other questions to see if he knows anything more than what v2.1.7 is stating (there was a later version in the writing some time back).

cbsiddharth, 2020-09-21:
Conversation:

> I had a quick look at the protocol document for File Transfer Messages
> and Multi-Part Message sections (there is also a glaring "Multi-Part
> Messages are not supported by OSDP" not sure what that is supposed to
> mean).
>
> I think this feature is underspecified; For instance, what command does
> it go under? osdp_FTcccc and osdp_FTRrrrr are not specified. Also,
> 
> - there is no way to set file attributes such as name before starting
> a transfer.
> - there is no indication of whether total file size can vary between
> messages (should PD error and discard?). Also if it cannot, why
> should it be sent over each packet?
> - a CP should be able to query file transfer progress status if it went
> thru a power cycle. No such command exists.
> 
> I'm sure we can make some educated guesses but we won't be playing nice
> with other implementations. If you have more information on these, I'd
> love to have this feature.

schmida2, 2021-01-11:
I am using the freshly released OSDP Specification adopted by IEC: IEC 60839-11-5 Edition 1.0 (2020-07).
And there I see the command osdp_FILETRANSFER along with its response osdp_FTSTAT.

The FILETRANSFER is not basing on the "Multi-Part Messages", this is independent. 

Is there already development ongoing for this feature request ?

cbsiddharth, 2021-01-11:
I very recently got my hands on the IEC spec, haven't haven't gotten arround to this yet.

If you are working on this, patches are welcome. :)

schmida2, 2021-06-22:
Hello @sidcha, I noticed your skeleton on file transfer addition, the code looks very nice again. 
Sure, "skeleton" means: it is not finished (e.g. file_tx_state_e is not used yet). Are you in the background further working on filertansfer? Or would you rather expect contribution?

sidcha, 2021-06-22:
I started working on it and got distracted; I do have some uncommitted changes. But if someone else beats me with a PR, I wont be too upset about it :)

At this point, it's about another day's worth of effort to get everything implemented and tested.

schmida2, 2021-06-25:
What would really help me better understand would be a description of the file operations. Is this API pretty stable already?

sidcha, 2021-06-25:
It’s been a while since I had a look at it. Let me check and add some code/comments to help understand it better.

closed 2021-06-27 by sidcha

jaldeepkaneria, 2024-03-13:
Hello guys, 
So i am using your library I have questions regarding multi-part message. But i want to add BIOREAD and BIOMATCH command i have added them in my local environment but as by default message buffer size is of only 256 bytes how to send more then 256 bytes in chunks i have read document from SIA version 2.2 then have given some insite aboute multi-part message but how to do it in any specific command?

commit bbdf4fc1 (2021-06-27): libosdp: Finalize file transfer support

This patch finalizes a basic version of file transfer support. This
implementation will probably fail in a dozen spectacular ways but for
the most straightforward cases it would work.

Closes: #24