osdp-dev/libosdp #25: Python import error, undefined symbol: EVP_DecryptFinal_ex

silchas, 2020-09-21:
Greetings.

I'm having some problems with getting the python module to work.

It builds and installs with no problem, but when i try to run following in python:
```
python3 -c "import osdp; print(\"pyosdp\", \"Version:\", osdp.get_version());"

Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: /usr/local/lib/python3.6/dist-packages/osdp-1.3.0-py3.6-linux-x86_64.egg/osdp.cpython-36m-x86_64-linux-gnu.so: undefined symbol: EVP_DecryptFinal_ex
```
Any clue to what might cause this error? 

This is on a fresh ubuntu 18.04 install, with the following done to make it build:
sudo apt-get install cmake
sudo apt-get install build-essential
sudo apt-get install libssl-dev
sudo apt-get install python3-sphinx
sudo apt-get install python3.6-dev
sudo apt-get install python3-pip
pip3 install setuptools

cbsiddharth, 2020-09-21:
Looks like pyosdp isn't linking to openssl correctly. The most recent commit is causing this regression, I'll fix this in a bit.

In the meantime, you could `git reset HEAD~1 --hard` get rid of this issue.

silchas, 2020-09-21:
> Looks like pyosdp isn't linking to openssl correctly. The most recent commit is causing this regression, I'll fix this in a bit.
> 
> In the meantime, you could `git reset HEAD~1 --hard` get rid of this issue.

That was fast, and it did the trick, thanks!

commit 4ece08d1 (2020-09-23): Allow LibOSDP to export its build rules to other modules

Fixes: #25
closed 2020-09-23 by cbsiddharth