osdp-dev/libosdp #27: Communication error when CP doesn't send 0xFF as the first character

z8a, 2020-09-24:
I don't know if the 0xFF as first character is mandatory but the access control I'm using as CP sends it only after it has received it from the DP. I've resolved the problem adding it at the beginning of the buffer when the length of received buffer is 0 and the first character is not 0xFF.
I've done this also in an old library release.

cbsiddharth, 2020-09-24:
The spec does mention that there must be 1 byte-width-time of channel inactivity between packets and the first byte of a packet must be 0xFF.

Nevertheless, I've seen some devices that don't follow this strictly. I'll add a compile time macro to disable sending/expecting the mark byte.

cbsiddharth, 2020-09-25:
You can now pass `-DCONFIG_OSDP_SKIP_MARK_BYTE=on` to cmake to disable the leading 0xFF sent/expected in all packets.

commit 1225ab03 (2020-09-25): Add compile time switch CONFIG_OSDP_SKIP_MARK_BYTE

Some PD/CP don't send the initial mark byte (0xFF) as defined in OSDP
specification. This patch adds a compile time switch to disable
sending/expecting the mark byte so LibOSDP can work with such devices.

Fixes: #27
closed 2020-09-25 by cbsiddharth