osdp-dev/libosdp #28: Support multi-drop on same channel

te-johan, 2020-09-26:
OSDP is a multi-drop protocol designed for RS-485 where multiple PD can be connected to the same two wires.
The CP part of libosdp does not support running multiple PDs over the same interface since the channel is unique for each PD.
I don't know how this library is and will be used but I'm guessing most people will have 1 CP and 1 PD. If they have multiple PD they will likely use the same interface.

I see two solutions to supporting multiple PDs on the same channel.

1. Change so only one channel is supported. This simplifies the code but it might be harder to emulate multiple PDs using message queues and similar.
2. Add a channel ID so that libosdp knows if two PDs share the same communication channel.

For both these solution an ownership token have to be used to determine which PD that owns the communication channel.
This can be implemented in cp_phy_state_update() where OSDP_CP_ERR_INPROG can be returned if the channel is busy and if it is not and osdp_cmd_dequeue returns a command to send out it should take the ownership of the channel. The channel ownership is dropped when a reply is received or when an error occurs.

Because of the design of osdp_cp_refresh() each PD will get equal "air time". So no channel scheduling had to be done.

cbsiddharth, 2020-09-26:
Thanks for the writeup, I like the channel ID approach. `utils/channel.c` can already handle multiple requests to open the same channel so I suppose, it can have an API to return channel ID too (the hashmap's hash_t itself perhaps).

cbsiddharth, 2020-09-27:
I don't have a hardware setup to test this right away but it should work (mostly).

closed 2020-09-27 by cbsiddharth

te-johan, 2020-09-27:
Well done, I’ll probably test it out during the week.  Maybe something like this can be used to emulate multiple PDs on a single line. https://github.com/rkubik/serial-proxy-c

commit caf7618f (2020-09-27): libosdp: Add support for multiple PDs connected in a single channel

This patch adds support for multiple PDs connected to a CP over a single
channel. A new channel->id is introduced to uniquely identify a channel
across connected PDs and each PD holds that channel until it's finished
with it.

Fixes: #28