osdp-dev/libosdp #40: Porting libosdp to FreeRTOS

atandersson, 2020-12-17:
Would it be possible to port this library for using with FreeRTOS running on ARM MCU's?
Some embedded solutions which are looking to start implementing OSDP today are running task schedulers like FreeRTOS on small ARM Cortex-M0/M3/M4 cores.
I think it would be a nice feature to include an option/fork for building the library for these devices as well, especially the PD's which are most likely to run on cheaper MCU's.

patrick-compass, 2021-01-18:
I'm currently trying to implement this library to operate a CP on a Maxim Cortex-M4. Any tips would be appreciated.

cbsiddharth, 2021-01-18:
Most of the platform dependent code is well abstracted, so you should be able to port libosdp to FreeRTOS with minimal effort. I have listed the things you need to do on the CP side, if you need to get the PD also working, similar methods exists for the PD too.

In `src/osdp_cp.c`:
- Read libosdp's [API documentation](https://libosdp.gotomain.io/api/control-panel.html) (probably outdated; yet useful).
- Setup an ISR to send/receive UART data using FreeRTOS's API.
- Modify `osdp_cp_setup()` with some ugly static stuffs that we always do in the embedded world.
- Setup a thread using FreeRTOS's API. In this thread, create the CP (using `osdp_cp_setup()`) and then run the main state loop (`osdp_cp_refresh()`) it whatever frequency you thing is reasonable (typically, one calls every 50ms is expected to meet various OSDP timing requirements).
- Modify `cp_send_command()` to send the OSDP packet over the ISR instead of calling `pd->channel.send()`.
- Modify `cp_process_reply()` so that the UART data can be notified (since we have and ISR now) instead of polling call to `pd->channel.recv()`.

In `src/osdp_common.c`:
- Implement SC methods `osdp_encrypt()`, `osdp_decrypt()` and `osdp_get_rand()` using some crypto library available in FreeRTOS.
- Implement `osdp_millis_now()` using some tick source provided by FreeRTOS so libosdp can get a sense of real world timings.

Hopefully, I've mentioned everything needed to accomplish this task do let me know if I missed anything. I did the above steps (and some more) when porting this library to Zephyr RTOS; you can have a look at [that implementation](https://github.com/zephyrproject-rtos/zephyr/tree/master/subsys/mgmt/osdp) if you need inspiration. Be informed that the zephyr implementation is not yet updated to libosdp's master.

Lastly, if you do get everything to work, don't forget to send a PR this way :-).

Cheers!

cbsiddharth, 2021-02-28:
@patrick-compass confirmed (by email) that he is able to run LibOSDP on FreeRTOS after redefining some internal methods locally. Those symbols have been marked weak with e3eb1fbd49c1bb189642c4230971232148e15bc8 so this can be done without any modification to the library.

Closing this issue.

closed 2021-02-28 by cbsiddharth

laszlopataki, 2022-01-12:
@sidcha The repo has quite changed since the last comment. 
Could you please give some updated tips for the porting? 
Thank you