osdp-dev/libosdp #42: Question on secure mode and library usage

prxvm, 2021-01-28:
First off, great work! I have some questions on the usage of this library, hope you can clarify:
- There is a setup flag to enable install mode on a PD. But say the PD is operational, how to get it into install mode? From the code, it doesn't look like it honors this flag during runtime, but only during initial setup. Is this expected to be handled by the application? Or should the init happen again?
- Second question is somewhat related to the first point. Before entering install mode, or maybe in some other situation, how can a PD reset communication with the CP so as to restart secure channel handshake? I saw this note from the the author on one of the GH issues saying one of the ways to handle from the CP, can you point me to the point in the CP implementation where this happens -
```
On a side note, if you are developing the CP internally, you can force the sequence number to 0 in a message that tries to restart communication to force the PD to discard the current SC session. LibOSDP's CP implementation does this and it has worked very well for me.
``` 
- Currently the PD setup sets the SC_CAPABLE flag to enable secure channel during init. This doesn't seem to be configurable from the application. Is there a way/API with which an application can set this flag during initialization to work in non-secure mode? It could potentially be trivially done by adding an arg in the setup function, but I was wondering if there is an existing way.

cbsiddharth, 2021-01-28:
> There is a setup flag to enable install mode on a PD.

The install-mode is an insecure mode (it uses SCBK_D and allowed any CP to set a key) and cannot and should not be allowed during normal operation of the PD. Enabling this is mode is up to the implementation and can be done by setting flag `OSDP_FLAG_INSTALL_MODE` in the `osdp_pd_info_t ::flags`. 

For instance, HID has some special configuration cards that you can present to the RFID reader soon after boot to force the device into install-mode. You could also have a physical button (that you cannot accidentally press) which you press and hold while restarting the device.

> Before entering install mode, or maybe in some other situation, how can a PD reset communication with the CP so as to restart secure channel handshake?

Typically, install-mode happens after a reboot that is covered. In other cases, since PD effectively owns the SC, it can just reply with a NAK and error code `OSDP_PD_NAK_SC_COND` and the CP would be forced to discard the current SC session and reinitiate the handshake. Since is is not a general workflow, LibOSDP does not allow you to do this.

> can you point me to the point in the CP implementation where this happens

Look here: https://github.com/goToMain/libosdp/blob/master/src/osdp_phy.c#L334

> Currently the PD setup sets the SC_CAPABLE flag to enable secure channel during init

Since LibOSDP has a working implementation of the secure channel, it advertises this as an implicit capability. I had a macro to disable secure channel in the past but then removed it because it beats the purpose of OSDP being a secure protocol. If for some reason you don't want to use it, you can call `osdp_pd_set_capabilities()` after `osdp_setup()` to set the compliance level of `OSDP_PD_CAP_COMMUNICATION_SECURITY` to 0.

> if I enable the keyset command and send it to the PD with some 16byte data, the PD still receives the same SCBK made before with the the master key and the info parameters

This should not happen, let me check.

prxvm, 2021-01-28:
Thank you for all the answers!
As for the last point on keyset command, I'd like to point out `osdp_cp_send_command_keyset` where the correct data gets copied into the keyset cmd and enqueued. But when `cp_build_command` gets called, it simply calls `osdp_compute_scbk` under the `CMD_KEYSET`  case. Not sure if this is intentional to keep the key computation per spec. Is sending an arbitrary key supported? If yes, then I guess `CMD_KEYSET` case under `cp_build_command` should check for data in keyset struct that was populated by `osdp_cp_send_command_keyset` function.

cbsiddharth, 2021-01-29:
So.. it looks like this part of the library hasn't received much attention since beginning of time, ergo missing a critical code path :). 

CP applications cannot set an arbitrary SCBK to connected PDs but only choose to change its own master key. When a master key change is requested, LibOSDP should prepare suitable SCBKs and send it to the corresponding PDs which have an active SC. I'll look into this tomorrow.

IDmachines, 2021-01-29:
Thanks for looking into this!

cbsiddharth, 2021-01-30:
@prxvm, @IDmachines, master key set issue is now fixed.

Do you have any further questions?

prxvm, 2021-02-01:
Good for now. Thanks for your quick feedback and commit!

closed 2021-02-01 by prxvm