osdp-dev/libosdp #43: osdp_phy_check_packet return value for incorrect PD

prxvm, 2021-02-01:
Return value of `osdp_phy_check_packet` in case a PD receives command poll command that is not intended it for i.e. meant for a different PD, shouldn't the following code block check for `ISSET_FLAG(pd, PD_FLAG_PD_MODE)` instead of `!ISSET_FLAG(pd, PD_FLAG_PD_MODE)` ?

```
  /* validate PD address */
  pd_addr = pkt->pd_address & 0x7F;
  if (pd_addr != pd->address && pd_addr != 0x7F) {
    /* not addressed to us and was not broadcasted */
    if (!ISSET_FLAG(pd, PD_FLAG_PD_MODE)) {
      OSDP_LOG_ERROR("Invalid pd address %d", pd_addr);
      return OSDP_ERR_PKT_FMT;
    }
    return OSDP_ERR_PKT_SKIP;
  }
```

cbsiddharth, 2021-02-01:
In PD mode, in multi-drop channels, when a PD receives a command addressed to another PD, it should not process the command: ie, should skip it and hence returns `OSDP_ERR_PKT_SKIP`.

On the other hand, a CP receiving a response from another PD when the channel is locked to it is treated as an error. This can happen only if a faulty PD is responding when it shouldn't.

prxvm, 2021-02-02:
Thank you for the clarification!

closed 2021-02-02 by prxvm