osdp-dev/libosdp #49: PD behaviour in secure mode for commands with no defined command structure

prxvm, 2021-02-25:
**Describe the bug**
I'm not sure if this is a bug but I'd like to know your thoughts. The secure session get discarded due to failed decryption upon receiving commands that have no command structure defined in the spec. Such commands are LSTAT, OSTAT, RSTAT, ISTAT. They simply instruct the PD to reply based on the command code. 

**Expected behavior**
Secure mode should not be discarded

**Observed behavior**
It is observed that secure session gets discarded when PD receives CMD_LSTAT in secure mode, due to decrypt function `osdp_decrypt_data` returning 0.

**Additional context**
Looking at the spec, the commands LSTAT, OSTAT, RSTAT, ISTAT have no command structure (i.e. no data to go along with the command), and could potentially have no decrypted data to work with. I think this is why returned length from `osdp_decrypt_data` is 0. Because the function considers 0 return value to be a failure scenario, this causes a secure session reset.

cbsiddharth, 2021-02-27:
> To be more specific, the decrypted data block for LSTAT looks like 80000000000000000000000000000000 which indicates that there is no data available, just the padding 0x80 to indicate end of message. So I don't think it is necessary to fail here.

@prxvm, According to the specification (v2.1.6, section D.3.2.1 and D.3.2.2) if a command or reply has no data, it should use SCS_15/SCS_16 and not SCS_17/SCS_18. You can see that LibOSDP handles this correctly while [sending](https://github.com/goToMain/libosdp/blob/master/src/osdp_phy.c#L446) and [receiving](https://github.com/goToMain/libosdp/blob/master/src/osdp_phy.c#L184).

Is this a commercially available PD that has this issue? If that is the case and it is widely used, we add a macro switch to allow this.

closed 2021-02-28 by cbsiddharth

prxvm, 2021-03-02:
@cbsiddharth this was observed a commercial panel. you are right this packet should not have been sent with SCS_17, but alas. Thank you.

commit aaba129f (2021-02-28): phy: Allow non-conformant, 0 length, encrypted data blocks

If command or reply has no data, PD "should" use secure message types
SCS_15 or SCS_16. But some PD seem to not implement this correctly. We
will be tolerant towards those faulty implementations.

Closes: #49