osdp-dev/libosdp #52: incompatibility with secure channel keys?

rsgmodelworks, 2021-04-15:
we are getting reports this code is not compatible with other OSDP implementations because it uses "MK" key derivation, which is not used in modern OSDP.

cbsiddharth, 2021-04-17:
@rsgmodelworks, thanks for looking into this.

> this code is not compatible with other OSDP implementations

This is not accurate. Reading the latest IEC spec, I believe this library should work as the _"MK" key derivation_ is not deprecated yet (correct me if I am wrong here).

This project was developed with v2.1.6 of the OSDP specification which does not dissuade the use of a shared master key to setup the SCBKs. That said, I'll add this topic to a list things I need to fix before bumping the OSDP supported version.

cbsiddharth, 2021-04-17:
Closing this as issue as it will be tracked as part of #53.

closed 2021-04-17 by cbsiddharth