osdp-dev/libosdp #53: Bump OSDP specification supported version to IEC Edition 1.0

cbsiddharth, 2021-04-17:
LibOSDP release v1.5.0 (inclusive) is based on v2.1.6 of the OSDP Specification. A later version of OSDP (v2.1.7 or IEC Edition 1.0) was released and this is a tracking issue for all changes that were introduced in that version.

- [ ] Add support for additional baud rates
- [ ] Deprecate master key based SCBK derivation

schmida2, 2021-04-23:
Good to see the SCBK handling is being revisited. By the way: SIA OSDP Specification v2.2 is already released, and as you say the master key based SCBK derivation is deprecated therein.
So per my understanding 2 options would be left:
1. ACU uses its own key management (and sends the individual SCBKs to the PDs with osdp_KEYSET command)
2. SCBK distributed out-of band (e.g. using a config card; no use of osdp_KEYSET command)

I do not think the libosdp should implement the full key management (random keys, rolling keys, ...) as this is not detailed in the spec. But overall, supporting the listed 2 options (plus the already implemented master key option??) would be nice...

cbsiddharth, 2021-04-23:
Yes, but Both those options converge into one (from the library’s perspective): allow ACU applications to send SCBK directly for each PD.

closed 2021-11-28 by sidcha