osdp-dev/libosdp #61: osdpctl compile error

prxvm, 2021-06-25:
**Unable to compile on master**
Running into osdpctl build error on master. Please advise.

**Expected behavior**
Clean build

**Observed behavior**
Running into the following error - 

```
[ 70%] Building C object osdpctl/CMakeFiles/osdpctl.dir/cmd_start.c.o
/Users/libosdp/osdpctl/cmd_start.c:290:9: error: no member named 'scbk' in 'osdp_pd_info_t'
                info->scbk = NULL;
                ~~~~  ^
/Users/libosdp/osdpctl/cmd_start.c:307:13: error: no member named 'scbk' in 'osdp_pd_info_t'
                info_arr->scbk = scbk;
                ~~~~~~~~  ^
/Users/libosdp/osdpctl/cmd_start.c:308:37: error: too few arguments to function call, expected 2, have 1
                c->pd_ctx = osdp_pd_setup(info_arr);
                            ~~~~~~~~~~~~~         ^
/usr/local/include/osdp.h:642:1: note: 'osdp_pd_setup' declared here
osdp_t *osdp_pd_setup(osdp_pd_info_t * info, uint8_t *scbk);
^
3 errors generated.
make[2]: *** [osdpctl/CMakeFiles/osdpctl.dir/cmd_start.c.o] Error 1
make[1]: *** [osdpctl/CMakeFiles/osdpctl.dir/all] Error 2
make: *** [all] Error 2

```

sidcha, 2021-06-25:
@prxvm, I cannot imagine you landed up at this state in your repository. The only time this error can happen is when the osdpctl is newer than the rest of the repo.

Can you copy-paste the output of the following commands in the current state of you repo:
``` bash
git status
```
```
git branch -v
```

prxvm, 2021-06-25:
Yes definitely, 

```
$ git branch -v
* master c02d7d6 Merge pull request #58 from prxvm/ignore-other-pd

$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
```

I have also updated the submodule
```
remote: Enumerating objects: 3582, done.
remote: Counting objects: 100% (513/513), done.
remote: Compressing objects: 100% (302/302), done.
remote: Total 3582 (delta 320), reused 372 (delta 211), pack-reused 3069
Receiving objects: 100% (3582/3582), 1.06 MiB | 3.95 MiB/s, done.
Resolving deltas: 100% (2422/2422), done.
Submodule 'utils' (https://github.com/goToMain/c-utils.git) registered for path 'utils'
Cloning into '/Users/vmore/my_test/up_libosdp/utils'...
remote: Enumerating objects: 816, done.
remote: Counting objects: 100% (199/199), done.
remote: Compressing objects: 100% (106/106), done.
remote: Total 816 (delta 110), reused 156 (delta 79), pack-reused 617
Receiving objects: 100% (816/816), 2.26 MiB | 5.58 MiB/s, done.
Resolving deltas: 100% (485/485), done.
Submodule path 'utils': checked out '2447feb5f56e0ff9b01e4d0427df357347a925bf'
```
is `osdpctl` a submodule?

sidcha, 2021-06-26:
Unbelievable. The CI is running those exact same steps to succeed on master. More information please :)

 - Build machine kernel/OS info
 - Compiler version
 - Run `make clean && make VERBOSE=1` and post the log to https://pastebin.com/ and share link.

prxvm, 2021-06-26:
Okay I found what is happening, a copy of `osdp.h` from an older version ended up in my `usr/local/include` (don't know how, need to reproduce, I did not place it there). Because this older `osdp.h` in `/usr/local/include` preceeded the `osdp.h` from the library, compiler picked that up and complained. I will close this issue. Thank you for your help!

closed 2021-06-26 by prxvm