osdp-dev/libosdp #63: Cannot import osdp library from python3.9

qdx, 2021-07-07:
**Describe the bug**
After successfully building and installing the python library, error was encountered when trying to import the osdp library. 

**Expected behavior**
The osdp library should be imported without error.

**Observed behavior**
When importing the osdp library via `import osdp`, below error is encountered:
```
➜  build git:(15) ✗ python3.9
Python 3.9.5 (default, May  4 2021, 03:36:27)
[Clang 12.0.0 (clang-1200.0.32.29)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import osdp
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: dlopen(/Users/dexinqi/Library/Python/3.9/lib/python/site-packages/osdp-1.5.0-py3.9-macosx-11-x86_64.egg/osdp.cpython-39-darwin.so, 2): Symbol not found: _hash32_djb2
  Referenced from: /Users/dexinqi/Library/Python/3.9/lib/python/site-packages/osdp-1.5.0-py3.9-macosx-11-x86_64.egg/osdp.cpython-39-darwin.so
  Expected in: flat namespace
 in /Users/dexinqi/Library/Python/3.9/lib/python/site-packages/osdp-1.5.0-py3.9-macosx-11-x86_64.egg/osdp.cpython-39-darwin.so
```

**Additional context**
I'm on macOS 11.4
This issues is encountered in version `771eae5`, the current master.

sidcha, 2021-07-08:
@qdx, thanks for reporting the issue. the c-utils submodule when through some cleanup which has caused this issue. I've fixed it now.

sidcha, 2021-07-08:
Also added PyOSDP builds and import check to CI builds. This should not happen again, sorry about the hassle.

commit 7232ad55 (2021-07-08): pyosdp: Fix #63 python3 import error

c-utils submodule when through a re-org to accommodate the --static-pd
option which required pyosdp to compile some additional files which did
not happen at that time (and CI didn't catch it for us). Add those files
to fix pyosdp builds.
closed 2021-07-08 by sidcha