osdp-dev/libosdp #64: integrating into yocto.

mmohtashamirad, 2021-07-15:
Hello,
I'm trying to integrate the library into a yocto cmake project as it's instructed in user manual. the problem is that I'm able to compile the same project in ubuntu and manually cross compile it for the target. but when it comes to yocto bitbake, it fails because it's compiling the libosdp for the host, not for target. 
does anyone have any idea about it?

sidcha, 2021-07-17:
You’d have to export cross compilers before invoking cmake. Did you have a
look at
https://libosdp.gotomain.io/libosdp/cross-compiling.html?

Even though it’s not something that you can directly use, it’s a good
starting point on what the library expects in order to be cross compiled.

mmohtashamirad, 2021-07-17:
> You’d have to export cross compilers before invoking cmake. Did you have a look at https://libosdp.gotomain.io/libosdp/cross-compiling.html? Even though it’s not something that you can directly use, it’s a good starting point on what the library expects in order to be cross compiled.
> […](#)

Thanks for your answer, Sidcha.
I already checked all parts of the manual pages. 
In yocto, you don't need to set cross compiler. It's done automatically. Actually, the main project is compiled for target, but only libosdp is being compiled for host.

sidcha, 2021-07-17:
@mmohtashamirad, can you give me some link/steps/reproducer to reproduce the problem locally? Did you have to make some changes to LibOSDP to get to where you are?

I'm not super familiar with Yocto and and there isn't much to go on here :)

mmohtashamirad, 2021-07-19:
Sure, I will try to provide you some guidance alongside with needed materials to build yocto and compile a simple osdp application using your library here. it takes a while for me to do so. I'll let you know when it's ready. 
also about the question "Did you have to make some changes to LibOSDP to get to where you are?": I haven't made any changes to your source code yet. and I think the best would be not to make any change to your library to keep it intact.

sidcha, 2021-08-10:
Ping?

sidcha, 2021-08-28:
@mmohtashamirad, feel free to reopen when you have something I can look into.

closed 2021-08-28 by sidcha

xJurgen, 2021-09-09:
We had same issue with integrating libosdp in yocto.
If you have "recipe-sysroot-native" in line "Found OpenSSL" (in do_configure cmake log), you have to change it to just "recipe-sysroot"
We solved it by adding openssl in recipe dependencies and now everything compiles accordingly.

DEPENDS += "openssl"

Also we compile only libraries, so we added EXTRA_OECMAKE option in recipe aswell

EXTRA_OECMAKE += "-DCONFIG_OSDP_LIB_ONLY:BOOL=ON"

sidcha, 2021-09-09:
@xJurgen, Thanks for letting us know. Please feel free to raise a PR for any changes needed to the library to make it easy for integration into yocto.

If you think that your yocto recipe can be helpful for others, you can send a PR for that too. There a [misc directory](https://github.com/goToMain/libosdp/tree/master/misc) to have configuration files.