osdp-dev/libosdp #65: OSDPCTL PD Mode - Command slab allocation failed

dman66, 2021-08-04:
**Describe the bug**
Upgraded from Release 1.4.0 to Release 1.5.0 and ran a simple test with 2 instances of OSDPCTL talking to each other over RS-485. 

After initial exchange the CP side will always fail with Command slab allocation failed.

**Expected behavior**
CP Mode should not have Command slab allocation failed.

**Observed behavior**

OSDP: INFO : CP : CP setup complete
OSDP: PD[0]: Sent [10] =>
    0000  ff 53 00 09 00 04 61 00  c0 66                    |.S....a..f      |
OSDP: PD[0]: Received [21] =>
    0000  ff 53 80 14 00 04 45 99  00 00 01 01 d2 04 00 00  |.S....E.........|
    0016  00 00 10 00 57                                    |....W           |
OSDP: PD[0]: Sent [10] =>
    0000  ff 53 00 09 00 05 62 00  a3 04                    |.S....b...      |
OSDP: PD[0]: Received [1] =>
    0000  ff                                                |.               |
OSDP: PD[0]: Received [9] =>
    0000  ff 53 80 08 00 05 46 ae  ff                       |.S....F..       |
OSDP: ERROR: PD[0]: CP : Command slab allocation failed
OSDP: ERROR: PD[0]: CP : Command slab allocation failed
OSDP: ERROR: PD[0]: CP : Command slab allocation failed
OSDP: ERROR: PD[0]: CP : Command slab allocation failed
OSDP: ERROR: PD[0]: CP : Command slab allocation failed
OSDP: ERROR: PD[0]: CP : Command slab allocation failed
OSDP: ERROR: PD[0]: CP : Command slab allocation failed
OSDP: ERROR: PD[0]: CP : Command slab allocation failed
.
.

**Additional context**
Modified the PD side (osdp_pd.c) to not have any implicit capabilities (don't report we support secure channel) but otherwise the code is fresh build of Release 1.5.0.

Also tried building with the latest code and the same CP Mode Command slab allocation failed happens.

sidcha, 2021-08-04:
@dman66, Have you tried this on the most recent master? I remember fixing something around this area :)

We are long due for a new release from master (pending some ongoing testing work), so you can safely upgrade without too many regressions.

dman66, 2021-08-05:
Yes, that fixes the issue. When I mentioned I had already tried master I think that was probably over a month ago so I didn't have this fix.

Thanks!

closed 2021-08-05 by sidcha