osdp-dev/libosdp #69: Multiple PD’s in a single piece of code

vk2tds, 2021-08-30:
The basic issue is that I am wanting to write some code that interfaces to an alarm system with an OSDP bus, and have my code implement four PD’s. As far as I can tell, all the code assumes a single PD per instance. 

Running multiple instances of the code, or using multiple instances of the library could work I guess but I am assuming this would involve ‘connections’ to the serial port which depending on platform would be exclusive to a single connection.

Assuming this is an enhancement, I am not sure what the cleanest way to implement this would be. Thanks.

sidcha, 2021-08-31:
Each PD is tied to a `osdp_t *ctx` object; _no_ part of LibOSDP is assumes a single PD (except for the log level). So you are free to do,

```c
#define NUM_PDS 4

osdp_t *pd_ctx[NUM_PDS];

for (i = 0; i < NUM_PDS; i++) {
    pd_ctx[i] = osdp_pd_setup();
}
```

As a side benefit to this, if you have distinct channels for each PD, then you can run each PD in a separate thread.

vk2tds, 2021-08-31:
Thanks for this. As soon as I get the hardware I will check this out.

vk2tds, 2021-09-01:
I am still coming up to speed with the library, and I am having some issues with the how to set up the physical layer. I am probably going to want to write my PD code for Zephyr, but it might be Linux. Regardless, I really cannot work out how to extend the base C example PD code to interface to the hardware. 

I am guessing this will need, in the case of a UART, to reference channel_open(), channel_read() and others. It will also 
need to reference setting up the pd_send and pd_recv functions. 

I also went looking for the c-utils documentation, but the c-utils link on the page to documentation points to https://c-utils.gotomain.io/ which is 404.

Once again, thanks for your help.

sidcha, 2021-09-02:
> I am probably going to want to write my PD code for Zephyr

There is already a port of this library on Zephyr RTOS upstream. It has not received much attention lately but still fully functional.

> I really cannot work out how to extend the base C example PD code to interface to the hardware.

If you are using Linux, you can use the python bindings or the included OSDPCtl tool to setup your PD. If it is Zephyr, physical layer is already implemented for you. For some other platform (or if you want to integrate to your own Linux application)

sidcha, 2021-09-02:
> I also went looking for the c-utils documentation, but the c-utils link on the page to documentation points to https://c-utils.gotomain.io/ which is 404.

This is expected. Long back, I was about to setup a documentation page for it but got distracted. In that land code is the best documentation.

vk2tds, 2021-09-02:
Thanks for this. I will definitely check out Zephyr - it looks like an interesting project. It looks like it will do most of what I need.

closed 2021-09-08 by sidcha

vk2tds, 2021-10-09:
As a first pass, I have been working on compiling libospd for the STM32 Nucleo 64 using the Arduino toolchain with some degree of success as a PD. It is compiling and running, but alas, is not actually working. I am still working through where the issue is. My code is likely the issue in there as the existing device on the OSDP bus on my InnerRange alarm is going offline with some regularity when my code is running.

That as it is, I have an update to the pd_receive_and_process_command() function in ospd.c. What I have found, at least in my toolchain, is that often there are multiple packets in the buffer that need to be processed. This code appears to correctly extract them. I am not saying this is perfect, but it is seeming to work at the moment. 

I have also found a case where I get the following bitstream 0xFF 0x53 0x00 0xFF 0x53 and things stall until there is a timeout. I have a fix for this but suspect I have another underlying bug to work on.

As soon as I get something running I will release the code. It is SOOOO close. Thanks. 

---

```
	do {
		err = pd_decode_packet(pd, &len);
		if (err == OSDP_PD_ERR_NO_DATA) 
			break;
		LOG_ERR ("Removing %d bytes from buffer with len %d err %lx", len, pd->rx_buf_len, err);
		if (pd->rx_buf_len >= 6){
			if ((pd->rx_buf[0]==0xff) && (pd->rx_buf[1]==0x53)){
				LOG_ERR ("...Address=%d LEN=%d MCI=%x", pd->rx_buf[2], pd->rx_buf[3] + (256*pd->rx_buf[4]), pd->rx_buf[5]);
			}
		}
		/* We are done with the packet (error or not). Remove processed bytes */
		remaining = pd->rx_buf_len - len;
		if (remaining > 0) {
			memmove(pd->rx_buf, pd->rx_buf + len, remaining);
		}

		/**
		 * Store remaining length that needs to be processed.
		 * State machine will be updated accordingly.
		 */
		pd->rx_buf_len = remaining;
	//} while ( (err != 0xffffffff) && (err != 0xfffffffe));
	//} while ( (pd->rx_buf_len > 0) && (err != OSDP_ERR_PKT_WAIT) && (err != OSDP_ERR_PKT_FMT));
	} while ( (pd->rx_buf_len > 0) && (err != OSDP_ERR_PKT_WAIT) );
```

Just for information, where I am up to at the moment is that keys are exchanged, then we get CMD: ID(61) and send REPLY: PDID(45). Soon after that I get errors for sequence repeat and soon the devices attempt to handshake again. I will let you know as soon as I get anywhere. 

```Welcomne to the Thunderdome...
OSDP: WARN : PD[14]: PD: SCBK not provided. PD is in INSTALL_MODE
OSDP: PD: Setup complete - @PROJECT_NAME@-@PROJECT_VERSION@ @GIT_BRANCH@ (@GIT_TAG@)
OSDP: PD[14]: Received pd_receive_and_process_command [7] =>
    0000  ff 53 05 09 00 04 62                              |.S....b         |
OSDP: PD[14]: Received pd_receive_and_process_command [a] =>
    0000  ff 53 05 09 00 04 62 00  92 70                    |.S....b..p      |
OSDP: ERROR: PD[14]: PD: Removing 10 bytes from buffer with len 10 err fffffffc
OSDP: ERROR: PD[14]: PD: ...Address=5 LEN=9 MCI=4
OSDP: PD[14]: Received pd_receive_and_process_command [7] =>
    0000  ff 53 0c 09 00 04 62                              |.S....b         |
OSDP: PD[14]: Received pd_receive_and_process_command [a] =>
    0000  ff 53 0c 09 00 04 62 00  70 38                    |.S....b.p8      |
OSDP: ERROR: PD[14]: PD: Removing 10 bytes from buffer with len 10 err fffffffc
OSDP: ERROR: PD[14]: PD: ...Address=12 LEN=9 MCI=4
OSDP: PD[14]: Received pd_receive_and_process_command [7] =>
    0000  ff 53 0e 09 00 04 62                              |.S....b         |
OSDP: PD[14]: Received pd_receive_and_process_command [a] =>
    0000  ff 53 0e 09 00 04 62 00  30 b3                    |.S....b.0.      |
OSDP: ERROR: PD[14]: PHY: None vk2tds
OSDP: CMD: CAP(62) [1] =>
    0000  62                                                |b               |
OSDP: DEBUG: PD[14]: PD: CMD: CAP(62) REPLY: PDCAP(46)
OSDP: ERROR: PD[14]: PD: Removing 10 bytes from buffer with len 10 err 0
OSDP: ERROR: PD[14]: PD: ...Address=14 LEN=9 MCI=4
OSDP: REPLY: PDCAP(46) [1b] =>
    0000  01 01 01 02 01 01 03 02  01 04 01 01 05 01 01 08  |................|
    0010  01 00 09 01 00 0a 00 02  0d 01 01                 |...........     |
OSDP: PD[14]: Received pd_receive_and_process_command [7] =>
    0000  ff 53 0e 13 00 0d 03                              |.S.....         |
OSDP: PD[14]: Received pd_receive_and_process_command [14] =>
[... 109 lines cut ...]
    0000  ff 53 03 08 00 00 61 00  41                       |.S....a.A       |
OSDP: ERROR: PD[14]: PD: Removing 9 bytes from buffer with len 9 err fffffffc
OSDP: ERROR: PD[14]: PD: ...Address=3 LEN=8 MCI=0
OSDP: PD[14]: Received pd_receive_and_process_command [7] =>
    0000  ff 53 03 08 00 00 61                              |.S....a         |
OSDP: PD[14]: Received pd_receive_and_process_command [9] =>
    0000  ff 53 03 08 00 00 61 00  41                       |.S....a.A       |
OSDP: ERROR: PD[14]: PD: Removing 9 bytes from buffer with len 9 err fffffffc
OSDP: ERROR: PD[14]: PD: ...Address=3 LEN=8 MCI=0
OSDP: PD[14]: Received pd_receive_and_process_command [7] =>
    0000  ff 53 03 08 00 00 61                              |.S....a         |
OSDP: PD[14]: Received pd_receive_and_process_command [9] =>
    0000  ff 53 03 08 00 00 61 00  41                       |.S....a.A       |
OSDP: ERROR: PD[14]: PD: Removing 9 bytes from buffer with len 9 err fffffffc
OSDP: ERROR: PD[14]: PD: ...Address=3 LEN=8 MCI=0
OSDP: PD[14]: Received pd_receive_and_process_command [7] =>
    0000  ff 53 03 08 00 00 61                              |.S....a         |
[... cut ...] and [9] =>
    0000  ff 53 03 08 00 00 61 00  41                       |.S....a.A       |
OSDP: ERROR: PD[14]: PD: Removing 9 bytes from buffer with len 9 err fffffffc
OSDP: ERROR: PD[14]: PD: ...Address=3 LEN=8 MCI=0
OSDP: PD[14]: Received pd_receive_and_process_command [7] =>
    0000  ff 53 06 09 00 04 62                              |.S....b         |
OSDP: PD[14]: Received pd_receive_and_process_command [a] =>
    0000  ff 53 06 09 00 04 62 00  72 be                    |.S....b.r.      |```