osdp-dev/libosdp #70: Issues trying to use OSDP with an InnerRange Inception

vk2tds, 2021-10-10:
I am attempting to use LibOSDP with an InnerRange Exception alarm system. I have compiled the library, with some changes onto the STM32 using the Arduino ecosystem. 

The exact code I am running lives in the following repositories:
	* https://github.com/vk2tds/libosdp_pd
	* https://github.com/vk2tds/libosdp_arduino

The PD is configured in Install Mode (PD is in INSTALL_MODE)

The PD that I am creating has the ID number of 1. There is an existing device that has the ID number of 0. The full debug log can be found at the following link: (https://pastebin.com/6FgGWAar). Looking at the appropriate lines, it appears that a SC session is created, and then things stop working. Alas, I am not familiar enough with the internal workings of the protocol to know what should be going on. 

```OSDP: DEBUG: PD[1]: PD: CMD: CAP(62) REPLY: PDCAP(46)
OSDP: DEBUG: PD[1]: PD: CMD: CHLNG(76) REPLY: CCRYPT(76)
OSDP: DEBUG: PD[1]: PD: CMD: SCRYPT(77) REPLY: RMAC_I(78)
OSDP: WARN : PD[1]: PD: SC Active with SCBK-D
OSDP: DEBUG: PD[1]: PD: CMD: ID(61) REPLY: PDID(45)
OSDP: INFO : PD[1]: PD: PD SC session timeout!
OSDP: ERROR: PD[1]: PHY: seq-repeat/reply-resend not supported!
OSDP: ERROR: PD[1]: PHY: seq-repeat/reply-resend not supported!
OSDP: ERROR: PD[1]: PD: CMD receive error/timeout - err:-1
```
The pastebin was created using the following PD_CAP
```    {
      .function_code = OSDP_PD_CAP_COMMUNICATION_SECURITY,
      .compliance_level = 0, // Try 1 later
      // more doc/libosdp/secure-channel.rst
      .num_items = 0
    },
```
NOTE: I also tried with compliance_level=1. The capture can be found at https://pastebin.com/0guBD9zy

Any hints in getting this working would be appreciated. Thanks.

sidcha, 2021-10-10:
From the logs, the CP sent you a CMD_MFG:
```
OSDP: PD[1]: Received [1f] =>
    0000  ff 53 01 1e 00 0d 02 17  80 af 0b 8a 7d 2d d9 82  |.S..........}-..|
    0010  c7 df e0 2a 7c d4 b2 66  ce 32 38 ac d9 a7 fd     |...*|..f.28.... |
OSDP: CMD: MFG(80) [5] =>
    0000  80 00 11 b9 10                                    |.....           |
```
For which your PD replied with:
```
OSDP: PD[1]: Sent [f] =>
    0000  ff 53 81 0e 00 0d 02 16  40 19 69 28 fb 69 a2     |.S......@.i(.i. |
```

Both those are valid OSDP packets. After this, the CP did not send anything for at least `OSDP_PD_SC_TIMEOUT_MS` (which causes the secure channel session to timeout with the message "OSDP: INFO : PD[1]: PD: PD SC session timeout!").

After this, things get interesting, your PD thinks it sees the next command but that's just the last command sitting in your rx buffer without being removed **plus** the new command. (notice that the first 31 bytes are identical to CMD_MFG) 
```
OSDP: PD[1]: Received [3f] =>
    0000  ff 53 01 1e 00 0d 02 17  80 af 0b 8a 7d 2d d9 82  |.S..........}-..|
    0010  c7 df e0 2a 7c d4 b2 66  ce 32 38 ac d9 a7 fd ff  |...*|..f.28.....|
    0020  53 01 1e 00 0d 02 17 80  af 0b 8a 7d 2d d9 82 c7  |S..........}-...|
    0030  df e0 2a 7c d4 b2 66 ce  32 38 ac d9 a7 fd ff     |..*|..f.28..... |
```
This caused LibOSDP PD to think that the CP is resending the same packet twice and that produces this error:
```
OSDP: ERROR: PD[1]: PHY: seq-repeat/reply-resend not supported!
```

Looks like you are causing this issue with your downstream changes. Can you remove any changes you made to pd_decode_packet() and report what errors you are seeing? In essence, what was the issue you faced that prompted you to patch pd_decode_packet()? -- that fix is not right and is causing this problem for you.

**nit:**
> OSDP: PD: Setup complete - @PROJECT_NAME@-@PROJECT_VERSION@ @GIT_BRANCH@ (@GIT_TAG@)

You need to run ./configure which will produce the `osdp_config.h` which you need to include in your project. Looks like you are including `osdp_config.h.in`

sidcha, 2021-10-21:
That should fix the sequence mismatch issue (your SC would still timeout as that is due to the CP's inaction). Please redo your tests with this fix and let me know if that works for you.

vk2tds, 2021-10-29:
Hi. You are **NOT** going to believe this. I **THINK** I have found what is going on. 

From the alarms datasheet:
```
Inception also features an RS-485 OSDP reader bus, meaning that up
to 8 Inner Range SIFER smart card readers, or 8 Wiegand readers via
OSDP <> Wiegand converters, can be connected directly to the
controller to provide card access for both in and out directions for all
doors
```

This sounds all nice and logical. They want you to use their hardware on the device. So, I made contact with the vendor. Their response was as follows

```
The Inception does not support any 3rd Party OSDP readers, so only SIFER readers can be used on the RS485 Reader port – this is the same for both the controller itself, as well as any connected SLAMs. The only option available for using 3rd Party OSDP readers may be via the OSDP<>Wiegand converter connecting into the Weigand port of a SLAM.
```

Essentially, they have implemented an OSDP port that by designs only operates with a superset of the OSDP (0x40 MFG) protocol, and will not operate with third party devices.

I think what is happening is that the OSDP software on the alarm is getting confused by an actual OSDP device on the bus, crashing and restarting. 

My next step is to look at writing a protocol decoder for OSDP. I have one of the manufacturers WIEGAND to OSDP converters. Assuming that I capture the entire transaction, including initial enrolment, I should be able to decode the entire transmission. This would allow me to see the MFG 0x40 packets and their responses. Once I decode these, I might be able to implement enough to talk OSDP to this device.

Once again, thanks for your help. I will keep you updated.

sidcha, 2021-10-30:
> The Inception does not support any 3rd Party OSDP readers, so only SIFER readers can be used on the RS485 Reader port

This doesn't make any sense to me at all :). If the idea is to have vendor locked devices, then why take the trouble of implementing OSPD - a protocol meant to promote interoperability - in the first place? For their sake, I hope this response is from a tech support engineer who doesn't know what he is taking about.

> My next step is to look at writing a protocol decoder for OSDP.

This is a nice idea; but you may have some trouble due to secure channel but that can be overcome easily.

vk2tds, 2021-10-30:
To be honest, I have a feeling that they want to leverage OSDP, but have commercial lock in. This is for their 'lower end' product. Their more expensive one does more explicitly support OSDP. Anyway, I have followed up and will report anything I get back.

I have started with the decoder, and am making progress. As the PD is not locked down, it appears to do a key exchange on power up, meaning that I can capture CMD_CHLNG/REPLY_CCRYPT/CMD_SCRYPT/REPLY_RMAC_I. I am now working through the libosdp code to see how to decode packets. I am getting there with that, but I think it will be a task for another day.

closed 2021-12-18 by sidcha