osdp-dev/libosdp #73: Error building python related parts

besmit, 2022-01-12:
I'm running into an issue seemingly related to Python.

When issuing the command "make check" within my path "/home/bjorn/libosdp/build" it fails with the following error:
[100%] Running tests..
/bin/sh: 1: py.test: not found
make[3]: *** [pytest/CMakeFiles/check.dir/build.make:77: pytest/CMakeFiles/check] Error 127
make[2]: *** [CMakeFiles/Makefile2:464: pytest/CMakeFiles/check.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:471: pytest/CMakeFiles/check.dir/rule] Error 2
make: *** [Makefile:296: check] Error 2

I did a search on py.test but cannot find it in any of the subdirectories.

Another issue I run into is when executing the "make python" command in my path "/home/bjorn/libosdp/python/build" it fails with the following error:
Scanning dependencies of target python
/usr/lib/python3/dist-packages/setuptools/dist.py:475: UserWarning: The version specified ('') is an invalid version, this may not work as expected with newer versions of setuptools, pip, and PyPI. Please see PEP 440 for more details.
  warnings.warn(
running build
running build_ext
building 'osdp' extension
error: unknown file type '' (from '')
make[3]: *** [CMakeFiles/python.dir/build.make:76: CMakeFiles/python] Error 1
make[2]: *** [CMakeFiles/Makefile2:235: CMakeFiles/python.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:242: CMakeFiles/python.dir/rule] Error 2
make: *** [Makefile:202: python] Error 2

Has something changed with the python directories/files?

Thanks,

Bjorn

sidcha, 2022-01-13:
Hey Bjorn, sorry of the trouble, yes a lot has changed in the testing area; not so much in the python area.

> I did a search on py.test but cannot find it in any of the subdirectories.

pytest is a python testing framework. I was using my own makeshift testing suite to test this project from its inception but then switched to pytest to leverage some of the niceness it provides and possible future growth. I have updated the top-level README.md to include instructions on how to run the tests, let me know if that helps.

> Has something changed with the python directories/files?

No, I don't think so. My guess is you were inside the `[repo-root]/python/build` directory and your ran a `cmake ..` instead of `cmake ../..`. The idea is, the path passed to `cmake` as the second argument should refer to the `[repo-root]` (in your case, `/home/bjorn/libosdp/`).

If that was not the issue, can you nuke the build dir fully and redo the whole thing and paste the full output (including the commands that you typed) so I can try to figure out what is going on?

besmit, 2022-01-13:
Hi Sidcha,

Tried your pointers which fixed the issue. About the second question you mention to use `cmake ../..` instead of  `cmake ..` I think the readme in the python folder has to be changed as it mentions `cmake ..`

/Bjorn

sidcha, 2022-01-13:
@besmit, most users just have one top-level build directory and work from there (as is the convention for cmake projects). I agree the readme in `python/` was misleading which is why I added a point there to indicate that the commands have to be run from repo-root (diff below).

``` diff
diff --git a/python/README.md b/python/README.md
index 56d03a0..110eec8 100644
--- a/python/README.md
+++ b/python/README.md
@@ -10,7 +10,7 @@ A python module `osdp` and classes `ControlPanel` and `PeripheralDevice` are
 made available when the sources of this directory is built and installed.

 After cloning this repo (with `--recurse-submodules`), you can do the following
-to build the python module.
+(from the repo root) to build the python module.

 ```bash
 mkdir build && cd build
```

closed 2022-01-13 by sidcha