osdp-dev/libosdp #76: Build error with Python Bindings on ARMv7 device

carlwain74, 2022-03-02:
THIS IS A TEMPELTE
------------------
Please **remove** sections that you are not filling data into (such as this one)

**Describe the bug**
Cannot build the osdp module

**Expected behavior**
Module should build

**Observed behavior**
First I tried `make python` but this is not a target.
Then I tried `make python_build` and I get the response below.

**Additional context**
Python 3.9.2 (Python 2.7.18 is also available)

**A Comprehensive Log file**
`/usr/lib/python3/dist-packages/setuptools/dist.py:475: UserWarning: The version specified ('') is an invalid version, this may not work as expected with newer versions of setuptools, pip, and PyPI. Please see PEP 440 for more details.
  warnings.warn(
running build_ext
building 'osdp' extension
error: unknown file type '' (from '')
make[3]: *** [CMakeFiles/python_build.dir/build.make:76: CMakeFiles/python_build] Error 1
make[2]: *** [CMakeFiles/Makefile2:235: CMakeFiles/python_build.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:242: CMakeFiles/python_build.dir/rule] Error 2`

[1]: https://libosdp.gotomain.io/libosdp/debugging.html

Let me know if there is anything else I need to supply

sidcha, 2022-03-02:
I just tried to run `make python_build` and it works well for me. Can you try these:
- pull the latest master
- nuke your build directory start over

If those doesn't help, please [pastebin](https://pastebin.com/) the output of the following for further investigation.
```
uname -a
gcc -v
python3 -V
cmake --version
make python_clean && make VERBOSE=1 python_build
```

carlwain74, 2022-03-03:
https://pastebin.com/jQXmXWZf

sidcha, 2022-03-03:
I tried building it on my Raspberry Pi 4, everything works fine for me (see logs [here](https://pastebin.com/89UKR6vz)). I used the Raspbian image [here](https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2022-01-28/).

```
pi@raspberrypi:~/work/libosdp/build $ cat /sys/firmware/devicetree/base/model; echo
Raspberry Pi 4 Model B Rev 1.1
pi@raspberrypi:~/work/libosdp/build $ uname -a
Linux raspberrypi 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux
```

The only difference is that you seem to be running a 32 bit Raspbian but that should not cause this issue. I think something else in your environment is causing it.

carlwain74, 2022-03-04:
So we were able to make progress by running the following commands

```
sudo apt install cmake python3 python3-pip python3-dev libssl-dev doxygen

git clone https://github.com/goToMain/libosdp --recurse-submodules

cd libosdp
mkdir build && cd build
cmake ..
make

make python_install
```

The make check failed, but I was able to at least fire up the sample apps. However, I have another issue and some questions so I'll close this and create a new issue.

carlwain74, 2022-06-02:
I would like to re-open this issue as I want to move beyond 1.5.0 and get running with 2.1.0. I have tried to build, but not getting anything built. Right now I'm using a python module called "osdp".

sidcha, 2022-06-02:
Can you help me with some more info on the kind of failure you are seeing?
Am I to assume you are still using the same build environment and your
previous resolution is no longer works?

carlwain74, 2022-06-03:
Sorry, after I reopened I did cloned a new repo and it magically worked. I have tested with both a CP and a PD against real HW and it works nicely.

closed 2022-06-03 by sidcha