osdp-dev/libosdp #78: Need help with sending/receiving commands

carlwain74, 2022-03-04:
So here is what I am trying to achieve

Write a PD emulator which talks to a real CP device. The goal is that we want to inject in different card formats and test our product.

The sample python code does not include POLL or RawData commands so I was unsure how this all fits in.

Up until now I had a python programing using this module (https://github.com/ryanhz/osdp-python), but it only acts as a CP, which is why I turned to this module so I can write both a CP and PD emulator. The current CP is receiving data from a physical reader Axis A8207 in insecure mode and being an old device probably only supports V1 spec.

In the samples I see a bunch of command structures which are used to transmit data. How does poll/card messages fall into this sample.

Sorry for so many questions, I'm new to OSDP and trying to ramp up my knowledge as fast as I can.

Also I hit a runtime error with the CP sample python script

```
pi@raspberrypi:~/git/libosdp/samples/python $ python3 cp_app.py
osdp: CP: INFO : Setup complete; PDs:1 Channels:1 - libosdp-1.5.0 master (d385ce4)
pyosdp Version: libosdp-1.5.0 Info: master (d385ce4)
osdp: CP: ERROR: PD101: Response timeout for CMD(60)
osdp: CP: ERROR: PD101: Response timeout for CMD(60)
Traceback (most recent call last):
  File "/home/pi/git/libosdp/samples/python/cp_app.py", line 114, in <module>
    main()
  File "/home/pi/git/libosdp/samples/python/cp_app.py", line 105, in main
    if (count % 100) == 99 and cp.is_sc_active(PD_0):
AttributeError: 'ControlPanel' object has no attribute 'is_sc_active'
```

carlwain74, 2022-03-04:
ok I removed the random code from the samples and I can see that the PD goes active.

`osdp: CP: INFO : PD101: SC Active`

Now I need to figure out how to send card data from my PD

sidcha, 2022-03-04:
> The sample python code does not include POLL or RawData commands so I was unsure how this all fits in.

POLL is a low level OSDP command that the python LibOSDP abstracts away for you using the high level `refresh()` method.
This is the same as you would do in a C application (calling the `osdp_cp_refresh()`) method.

You can send card data to the CP by creating an even and sending it to the CP using the `notify_event()` call (the python sample is already doing this).

More generally:
- CP sends _commands_ to PD
- PD sends _events_ to CP

The command part is an OSDP spec introduced concept. The event is a LibOSDP specific thing to signal stuffs back to the CP by responding with this event as a response to the next POLL from CP.

> ok I removed the random code from the samples and I can see that the PD goes active.

What random code?

carlwain74, 2022-03-04:
Many thanks for responding.. The random code was the code you added to randomly send commands. I'm not trying to send commands from CP, just events to the CP for my test infrastructure.

So I need to do two things

1) Generate event on the PD
2) Process the event on the CP

So I tried to send this event data and it crashed.

```
card_read_event = {
    "event": osdp.EVENT_CARDREAD,
    "reader_no": 1,
    "format": osdp.CARD_FMT_RAW_WIEGAND,
    "direction": 1,
    "data": bytes.fromhex('0F933200')
}
```

```
Sending {'event': 0, 'reader_no': 1, 'format': 1, 'direction': 1, 'data': b'\x0f\x932\x00'}
Traceback (most recent call last):
  File "/home/pi/git/libosdp/samples/python/pd_app.py", line 135, in <module>
    main()
  File "/home/pi/git/libosdp/samples/python/pd_app.py", line 123, in main
    pd.notify_event(events[r])
KeyError: "Key: 'length' of type: int expected"
```

Where is length calculated?

How does the data present itself on the CP side?

sidcha, 2022-03-04:
> Where is length calculated?

This is happening in [python/pyosdp_cmd.c  +487](https://github.com/goToMain/libosdp/blob/master/python/pyosdp_cmd.c#L487). When you set the card type to `CARD_FMT_RAW_WIEGAND` or `CARD_FMT_RAW_UNSPECIFIED`, you are also expected to set the length field in the event dictionary and it has to be the number of bits of data instead of bytes. OTOH, when specifying `CARD_FMT_ASCII`, you have to set it to the number of bytes.

> How does the data present itself on the CP side?

As normal card reads are presented to the CP: in response to the osdp_POLL command.

carlwain74, 2022-03-04:
Okay, now we are cooking

I'm getting this on the CP side

`Address:  0  Event:  {'event': 0, 'reader_no': 1, 'format': 1, 'direction': 0, 'length': 32, 'data': b'\x0f\x932\x00'}`

carlwain74, 2022-03-04:
Now I need to hook up my PD emulator to a CP and see how that works out. What control panels have you tested against?

sidcha, 2022-03-04:
I have tested the CP implementation agains a lot of commercial PDs (including ones from HID). The PD implementation has been tested against our own CP (by means of unit and integration tests); users of LibOSDP have tested it agains real CPs too.

carlwain74, 2022-03-04:
Do you know what version of the Spec you are compliant against.

sidcha, 2022-03-04:
The most recent one: OSDP version 2.2.

carlwain74, 2022-03-07:
We can close this issue, thanks for helping out.

closed 2022-03-07 by carlwain74