osdp-dev/libosdp #97: OSDP_CMD_ABORT is not exposed to the application.

surendersampath, 2022-12-08:
**Describe the bug**
CMD_ABORT's command ID was fixed in issue #93 . However, this is not being exposed to the application.
Hence, not able to invoke abort command via `osdp_cp_send_command()` .
[image]

**Expected behavior**
`CMD_ABORT` to be exposed to application.

**Observed behavior**
`CMD_ABORT` is not exposed to application.

sidcha, 2022-12-08:
@surendersampath, the OSDP defined command `CMD_ABORT` is not meant for external use by applications, libosdp abstracts those details and provides its own high level commands though `enum osdp_cmd_e`.

Something _is_ missing though: the file transfer API `osdp_file_tx_abort` should send CMD_ABORT to the peer before reseting the state. Do you have any other use case for why ABORT needs to be exposed to the application?

surendersampath, 2022-12-08:
I would like to use the abort command to stop the ongoing file transfer. But, Do you suggest that just calling file_tx_abort() function is sufficient?

sidcha, 2022-12-08:
No, I spoke too soon, `osdp_file_tx_abort ` is an internal method and servers a different purpose.

After some inspection, I think it more sensible to abort an ongoing transfer using the existing `OSDP_CMD_FILE_TX` command's flags (`osdp_cmd_file_tx::flags`) field. Will make that change.

sidcha, 2022-12-09:
@surendersampath, You can use the newly introduced bit `OSDP_CMD_FILE_TX_FLAG_CANCEL` in `osdp_cmd_file_tx::flags` to abort an ongoing transfer. PD will not receive this command directly (as with other commands); instead the CP will send OSDP command `CMD_ABORT` to PD to signal the premature termination of current file transfer.

There is also a test case to ensure its correctness, so I will close this issue. For the command build failure, I'll create a new issue to track it.

closed 2022-12-09 by sidcha