osdp-dev/libosdp #100: TX_Decode failing for File Transfer Abort at PD.

surendersampath, 2022-12-09:
**Describe the bug**
PD does not handle the incoming FILE TX command from CP with flag set as `OSDP_CMD_FILE_TX_FLAG_CANCEL` .
Decode fails as the length is lesser than expected.

[image]

**Expected behavior**
PD to handle the the file transfer command with `OSDP_CMD_FILE_TX_FLAG_CANCEL` .
**Observed behavior**
A description of what happened.

**Additional context**
I'm using the below code snipped to abort an on-going file transfer. 
```
// Function to Abort File Transfer
bool OsdpAbortFileTransfer(eOsdpFileTypes xFileType)
{
    bool xStatus = false;
    struct osdp_cmd xAbortCommand;
    xAbortCommand.id = OSDP_CMD_FILE_TX;
    xAbortCommand.file_tx.id = xFileType;
    xAbortCommand.file_tx.flags = OSDP_CMD_FILE_TX_FLAG_CANCEL  ;
    xStatus = ( osdp_cp_send_command( ctx, 0, &xAbortCommand ) == 0 ) ? true : false;

    return xStatus;
}
```

sidcha, 2022-12-10:
> PD does not handle the incoming FILE TX command from CP with flag set as `OSDP_CMD_FILE_TX_FLAG_CANCEL`.

`OSDP_CMD_FILE_TX_FLAG_CANCEL` is not sent to the PD, it is consumed by CP to make the decision of sending the ABORT command. When the CP receives a command that has the cancel bit set, and no transfer is in progress (or the file ID does not match) we [fail the command](https://github.com/goToMain/libosdp/blob/master/src/osdp_file.c#L290)!

So, I suspect something else is causing the failure. What do you need to do to reproduce this issue? The best thing to do is to write a unit test that can reproduce this issue (everything needed to write such a test can be found [here](https://github.com/goToMain/libosdp/blob/master/tests/pytest/test_file_tx.py)).

> Suggested FIX : Add file_state_reset(f); in the TX_Decode if any error(such as length mistmatch) occurs?

Since the failure is between LibOSDP CP <==> PD, I'd prefer to root cause and fix the bug properly rather than mask the issue by fixing the symptom with file_state_reset().

surendersampath, 2022-12-10:
Thanks for looking into this. I'll try and re-create this issue using your unit test set up.

sidcha, 2023-01-17:
@surendersampath, I have a potential fix for this issue. For now, I'm closing this issue but feel free to re-open if you can still reproduce it.

closed 2023-01-17 by sidcha