osdp-dev/libosdp #106: Secure Channel gets disconnected at seemingly random intervals

tim661811, 2023-03-03:
**Describe the bug**
We use the libosdp library to communicate to Peripheral Devices, so we act as the Control Panel. We use different kinds of OSDP readers (Idesco and STiD readers). The issue we're having is that the secure channel gets disconnected at seemingly random intervals (could be twice a minute, but sometimes it's also once every 15 mins). After the Secure Channel is lost, it gets rebuild within 1 second.

**Expected behavior**
 Secure connection should not be lost

**Observed behavior**
In case of the Idesco reader, all of a sudden libosdp sends an unsecure POLL event which the PD rejects with a NAK
[image]

The STiD behavior is slightly different, but the problem seems to occur in the same place. A secure POLL event is sent to the PD by libosdp and then rejected by the PD because the CRC is wrong
[image]

**A Comprehensive Log file**
OSDP sniffer output for Idesco situation:
[image]

OSDP sniffer output for STiD situation:
[image]
CRC should've been
[image]

sidcha, 2023-03-04:
@tim661811, Thanks for creating this comprehensive bug report. Can you also provide the logs provided by libOSDP with log level set to DEBUG and [packet trace enabled](https://libosdp.gotomain.io/libosdp/debugging.html#packet-trace-builds)? I cannot make out a whole lot of details form the screenshot you've shared.

The CP would send a POLL in plain text only if has already lost secure channel with the PD. So we need to backtrack and look at why it is loosing the secure channel.

tim661811, 2023-03-06:
Hereby the requested logs, I let both readers run for a few minutes and piped the results to a file.

[idesco_reader.log](https://github.com/goToMain/libosdp/files/10897513/idesco_reader.log)
[stid_reader.log](https://github.com/goToMain/libosdp/files/10897514/stid_reader.log)

sidcha, 2023-03-06:
Some log messages does not make sense with the control flow in code (order). Do you have some downstream modifications to the CP state machine? If you do, can you revert them and run on master with those readers and tell me if you can still reproduce this issue? (and of course if you can reproduce it on master, create log files as you have done now).

tim661811, 2023-03-07:
We did indeed alter the order of the state machine because STiD readers had an issue where the capability request could only be performed in Secure Channel mode. I reverted this change and ran libosdp again with the Idesco reader.

Resulting log:
[idesco_reader.log](https://github.com/goToMain/libosdp/files/10907003/idesco_reader.log)

sidcha, 2023-03-07:
The important part is this:
```
osdp: CP: WARN : RS485-1: Response timeout for CMD(60)
osdp: CP: WARN : RS485-1: PD replied with NAK(6) for CMD(60)
```
I should add timestamps to these logs soon, but it appears the CP marked that PD as offline (because the PD didn't respond within the timeout) and came back after sleeping for a while, between those two log lines. When coming back from sleep, the CP starts setting up a new SC session (and rightly so) but it appears the PD hasn't decarded the previous SC session yet so it responds with NAK(6).

This is a bug in the PD firmware (please report it to the vendor) as it should discard secure channel after a period of inactivity (of course, it cannot keep the session endlessly active). To workaround this problem for the time being, you can increase the response timeout to some higher value and debug why the response from the PD is taking that long (ie., find where time is spent most).

sidcha, 2023-03-07:
> STiD readers had an issue where the capability request could only be performed in Secure Channel mode

This is also wrong and a catch-22. The capability report is what the CP uses to know that a PD supports secure channel; so we cannot make that request in a secure channel. Please report this one too :)

tim661811, 2023-03-07:
Okay, interesting that two different manufacturers have the same issue. I'm gonna play around with the response timeout and see if that resolves the issue for now, and then get back to the manufacturers. 

The capability request was indeed flagged as an issue and has since been resolved in their newer firmware version, but at the time we couldn't wait for the fix, so we had to change the order of the libosdp state machine for now.

Thanks for the help!

closed 2023-03-07 by tim661811

MarCovy, 2023-06-29:
Hello, 
@tim661811, what is your program/application for analyzing the OSDP protocol? 
The screenshots from your first post look great!
Thanks for the help!

tim661811, 2023-06-29:
> Hello, @tim661811, what is your program/application for analyzing the OSDP protocol? The screenshots from your first post look great! Thanks for the help!

I used a tool called OSDPPro which is a tool made by OSDP reader manufacturer STiD. Unfortunately, I don't think this is a publicly available tool.

MarCovy, 2023-06-29:
@tim661811 yes, that's right, this is not public tool :)
Could you give my some advice about usage of this library, I want to start use Secure Channel between reader and my Raspberry Pi. I have this part of code:

    int sample_cp_send_func(void *data, uint8_t *buf, int len)
    {
        (void)(buf);
    
        QByteArray temp = QByteArray((char*)buf, len);
        LOG(INFO) << "Sended: " << temp.toHex();
        OSDPDriver* driver = (OSDPDriver*)data;
        driver->send(buf, len);
        return len;
    }

    int sample_cp_recv_func(void *data, uint8_t *buf, int len)
    {
        (void)(len);
        OSDPDriver* driver = (OSDPDriver*)data;
        QByteArray buffer = driver->recv();
        for (int i = 0; i < buffer.length(); i++) {
            *(buf+i) = buffer[i];
        }
    
        LOG(INFO) << "Recv: " << buffer.toHex();
        return buffer.length();
    }

    int event_callback(void *arg, int pd, struct osdp_event *ev) {
        LOG(INFO) << "Event: " << pd;
        LOG(INFO) << "Event type: " << ev->type;
        return 0;
    }
    uint8_t key[16] = {
        0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
        0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
    };

    osdp_pd_info_t pd_info[] = {
        {
            .name = "pd[101]",
            .baud_rate = 9600,
            .address = 0x00,
            .flags = OSDP_FLAG_ENFORCE_SECURE, //OSDP_FLAG_ENFORCE_SECURE
            .id = {},
            .cap = nullptr,
            .channel = {
                .data = this,
                .id = 0,
                .recv = sample_cp_recv_func,
                .send = sample_cp_send_func,
                .flush = nullptr
            },
            .scbk = key,
        }
    };

    cp.logger_init(OSDP_LOG_MAX_LEVEL, NULL);
    cp.setup(1, pd_info);
    cp.set_event_callback(event_callback, this);

    connect(&m_timer, &QTimer::timeout, [=](){
        cp.refresh();
    });

    m_timer.start(25);

Everythin is good without non secure channel, but i want to switch to secure protocol, but lib still use non secure option. Logs below:

    osdp: CP: INFO : Setup complete; PDs:1 Channels:1 - libosdp-2.2.0 master (6dc2e4a)
    osdp: CP: INFO : PD not online
    2023-02-01 18:26:14,481 INFO [default] Sended: ff530008000460ebaa
    2023-02-01 18:26:14,506 INFO [default] Recv: ff53800900044106c0d4
    osdp: CP: WARN : pd[101]: PD replied with NAK(6) for CMD(60)
    2023-02-01 18:26:14,553 INFO [default] Sended: ff53000900056100f051
    2023-02-01 18:26:14,577 INFO [default] Recv: ff538009000541011793
    osdp: CP: WARN : pd[101]: PD replied with NAK(1) for CMD(61)
    osdp: CP: DEBUG: pd[101]: CMD: ID(61) REPLY: NAK(41)
    osdp: CP: ERROR: pd[101]: Unexpected REPLY(41) for cmd CMD_ID
    2023-02-01 18:26:15,673 INFO [default] Sended: ff530008000460ebaa
    2023-02-01 18:26:15,698 INFO [default] Recv: ff53800900044106c0d4
    osdp: CP: WARN : pd[101]: PD replied with NAK(6) for CMD(60)

I was looking for some snipets in repo but I didn't find antything. Do you have some advice for me how to enable this secure option in this lib? Can you share some snippet? 

Thanks for the help!