osdp-dev/libosdp #109: Incorrect handling of length in CMD_ABORT

surendersampath, 2023-03-08:
**Describe the bug**
In the PD side, the length is handled incorrectly for a zero length command. CMD_ABORT is a zero length command just like CMD_POLL. Hence, its length should be checked similarly as it's being done in CMD_POLL.

**Expected behavior**
[image]
**Observed behavior**
[image]

[1]: https://libosdp.gotomain.io/libosdp/debugging.html

sidcha, 2023-03-08:
Thanks for reporting it.

Btw, you could have created a PR for this change in half the time it took to create this issue. I'm not complaining, just saying :).

closed 2023-03-08 by sidcha

surendersampath, 2023-03-08:
Thanks. Will do it next time. Cheers.

commit 3c5c9474 (2023-03-08): pd: Fix CMD_ABORT length check

Fixes: #109