osdp-dev/libosdp #125: Can't init Secure Channel

MarCovy, 2023-06-30:
Hello, I am using libosdp to connect to an Idesco reader with my Raspberry Pi. Everything is fine when I use a non-secure channel. I can ping my device, etc. Now, I want to switch to a secure channel for the connection. However, the library still wants to communicate with the reader using a non-secure channel. Is there any example or troubleshooting guide available for this case? Below, I have attached a part of my code and some fragments of the logs.

**Expected behavior**
A CP should sent firstly CMD_KEYSET 0x75 to set encryption key.

**Observed behavior**
Secure channel not starting.

**CODE**
```
int sample_cp_send_func(void *data, uint8_t *buf, int len)
{
    (void)(buf);

    QByteArray temp = QByteArray((char*)buf, len);
    LOG(INFO) << "Sended: " << temp.toHex();
    OSDPDriver* driver = (OSDPDriver*)data;
    driver->send(buf, len);
    return len;
}

int sample_cp_recv_func(void *data, uint8_t *buf, int len)
{
    (void)(len);
    OSDPDriver* driver = (OSDPDriver*)data;
    QByteArray buffer = driver->recv();
    for (int i = 0; i < buffer.length(); i++) {
        *(buf+i) = buffer[i];
    }

    LOG(INFO) << "Recv: " << buffer.toHex();
    return buffer.length();
}

int event_callback(void *arg, int pd, struct osdp_event *ev) {
    LOG(INFO) << "Event: " << pd;
    LOG(INFO) << "Event type: " << ev->type;
    return 0;
}

uint8_t key[16] = {
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
};

osdp_pd_info_t pd_info[] = {
    {
        .name = "pd[101]",
        .baud_rate = 9600,
        .address = 0x00,
        .flags = OSDP_FLAG_ENFORCE_SECURE, //OSDP_FLAG_ENFORCE_SECURE
        .id = {},
        .cap = nullptr,
        .channel = {
            .data = this,
            .id = 0,
            .recv = sample_cp_recv_func,
            .send = sample_cp_send_func,
            .flush = nullptr
        },
        .scbk = key,
    }
};

cp.logger_init(OSDP_LOG_MAX_LEVEL, NULL);
cp.setup(1, pd_info);
cp.set_event_callback(event_callback, this);

connect(&m_timer, &QTimer::timeout, [=](){
    cp.refresh();
});

m_timer.start(25);
```

**A Comprehensive Log file**
```
2023-02-02 17:03:38,678 INFO [default] Serial port connected
osdp: CP: INFO : Setup complete; PDs:1 Channels:1 - libosdp-2.2.0 master (6dc2e4a)
2023-02-02 17:03:43,170 INFO [default] Sended: ff530008000460ebaa
2023-02-02 17:03:43,195 INFO [default] Recv: ff53800900044106c0d4
2023-02-02 17:03:43,195 INFO [default] PD: CMD: 96
osdp: CP: WARN : pd[101]: PD replied with NAK(6) for CMD(60)
OSDP: CMD: ID(61) [1] =>
    0000  00                                                |.               |
P_TRACE_SEND: CP->PD[0]: [10] =>
    0000  ff 53 00 09 00 05 61 00  f0 51                    |.S....a..Q      |
2023-02-02 17:03:43,248 INFO [default] Sended: ff53000900056100f051
2023-02-02 17:03:43,273 INFO [default] Recv: ff538009000541011793
osdp: CP: WARN : pd[101]: PD replied with NAK(1) for CMD(61)
osdp: CP: DEBUG: pd[101]: CMD: ID(61) REPLY: NAK(41)
P_TRACE_RECV: PD[0]->CP: [9] =>
    0000  53 80 09 00 05 41 01 17  93                       |S....A...       |
OSDP: REPLY: NAK(41) [1] =>
    0000  41                                                |A               |
2023-02-02 17:03:43,273 INFO [default] PD: CMD: 97
osdp: CP: ERROR: pd[101]: Unexpected REPLY(41) for cmd CMD_ID
2023-02-02 17:03:44,373 INFO [default] Sended: ff530008000460ebaa
2023-02-02 17:03:44,398 INFO [default] Recv: ff53800900044106c0d4
2023-02-02 17:03:44,399 INFO [default] PD: CMD: 96
osdp: CP: WARN : pd[101]: PD replied with NAK(6) for CMD(60)
OSDP: CMD: ID(61) [1] =>
    0000  00                                                |.               |
P_TRACE_SEND: CP->PD[0]: [10] =>
    0000  ff 53 00 09 00 05 61 00  f0 51                    |.S....a..Q      |
```

sidcha, 2023-06-30:
Hi @MarCovy,

The key information is this:

> PD replied with NAK(6) for CMD(60)

Your PD responded with NAK error code 6. According to the spec this means "Communication security criteria not met". This error can be returned for a variety of reasons but in your case I think it is because you haven't set the SCBK to the PDs beforehand. LibOSDP will attempt to do this automatically (when you don't pass `OSDP_FLAG_ENFORCE_SECURE`) but the catch is that, for this to work, the PD has to be put to a special state called "install mode".

Different PDs have different ways of doing this (also, I have explained some of the popular methods in other issues in this repository) so you'd have to look it up in the manual. Also, have you had a chance to read our documentation in https://libosdp.gotomain.io/libosdp/secure-channel.html about this? If not, I'd suggest you do that too.

sidcha, 2023-06-30:
Closing this issue, as it is a question. But feel free to re-open the issue and/or keep the discussion going.

closed 2023-06-30 by sidcha