osdp-dev/libosdp #128: How to build libosdp for Android platform

HiLittlePiper, 2023-08-15:
Hello,

I want to compile this library for Android App, but I did not find any introduction out in the documentation, do you have any introduction to guide how to build Android library?

Thanks

sidcha, 2023-09-03:
I haven't done this but I don't think it should be a problem. It should be fairly straightforward to compile with Android's NDK build. Please follow the plain make build (./configure.sh && make) and adapt it to your Android.mk file.

closed 2023-09-03 by sidcha

sidcha, 2023-10-03:
@HiLittlePiper, have you had any luck with this? You can still use this issue page to ask help or post findings. Thought I am no expert on android NDK builds, I will try my best to help you with any libosdp specific issues you may have.