osdp-dev/libosdp #132: libOSDP without queues?

MatTab3335, 2023-10-03:
**Is your feature request related to a problem? Please describe.**
I want to run OSDP lib on STM32, but without using any RTOS, so the problem is in queues, is it possible somehow run this lib without queue.h ?

sidcha, 2023-10-03:
I don't see the connection, sorry; queue.h depends on list.h and both list.c and queue.c don't impose any OS specific requirements so you should be able to compile it to bare-metal. Can you please elaborate how queue.h is causing issues for you?

MatTab3335, 2023-10-03:
I'm sorry, instead of "slab", I wrote "queue". So the problem is to use slab functions and structures, I can't find anything similar to use with stm32

sidcha, 2023-10-03:
Slab also does not depend on anything OS specific.

Most of the modules in [c-utils repo](https://github.com/goToMain/c-utils/ ) is written to be OS independent. All you have to do is build a subset of that repository for STM32 and then build libosdp along with it.

MatTab3335, 2023-10-03:
Ok, thank you, I'll try

sidcha, 2023-10-03:
I will close this issue for now, please keep the conversation going if need be.

closed 2023-10-03 by sidcha

vk2tds, 2023-12-08:
I am currently running the library on an STM32F411 under Arduino, and it is working fairly well. https://github.com/vk2tds/libosdp_arduino