osdp-dev/libosdp #153: Issue with Keyset Command

FouxFox, 2024-01-03:
I have been beating on this for a while and can't seem to figure out where I'm going wrong.

I have this library fully functional when unencrypted, but when I go to use SC, it fails our after getting the RMAC_I. I poked around the code and found that for some reason ```cmd->keyset.type``` gets set to 50 between the initialization of the PD and the construction of the KEYSET command.

I checked the code files against master to ensure that these functions were not changes. I am using mbedtls.

### Log
```
Failed to verify PD Cryptogram
SC Failed. Retry with SCBK-D
CMD: CHLNG(76) Reply: CCRYPT 76)
CMD: SCRYPT(77) Reply: RMAC_I(78)
SC ACtive with SCBK-D. Set SCBK
Invalid key length
Failed to build packet for CMD(117)
```

### Code
```cpp
static const uint8_t key_data[16] = {
	0x20, 0x21, 0x22, 0x23, 0x24, 0x25, 0x26, 0x27,
	0x38, 0x39, 0x3A, 0x3B, 0x3C, 0x3D, 0x3E, 0x3F
};

osdp_pd_info_t pd_info[] = {
  {
    .baud_rate = 9600,
    .address = 0,
    .flags = 0,
    .channel = {
      .data = nullptr,
      .id = 0,
      .recv = *receiveData,
      .send = *sendData,
      .flush = nullptr
    },
    .scbk = key_data,
  },
};

cp.setup(1, pd_info);
```

### Trace
[OSDP_Trace_240103_151443.txt](https://github.com/goToMain/libosdp/files/13823486/OSDP_Trace_240103_151443.txt)

sidcha, 2024-01-09:
@FouxFox, I'd need more info than that to look into it (traces from libosdp, I don't understand the logs that you've attached).

Also, we have [tests](https://github.com/goToMain/libosdp/blob/master/tests/pytest/test_commands.py#L133C5-L133C24) to ensure that this command works as intended. Can you see if something is missing in this test case?

sidcha, 2024-01-14:
Closing this due to insufficient data. Please feel free to re-open if you have more inputs here.

closed 2024-01-14 by sidcha