osdp-dev/libosdp #157: CP SETKEY failing for PD in install mode

morcant, 2024-02-11:
I have a HID Signo 40 PD without keys set. It is in install mode with secure channel v2 enabled. 

libosdp appears to negotiate an SC with SCBK-D, but then KEYSET fails due to no SC.

I would expect after negotiating SC with SCBK-D, a KEYSET would succeed.

pyosdp: [2024-02-11T19:05:05] [INFO ] src/osdp_cp.c:1360: Setup complete; LibOSDP-2.4.0 master (77c3ba0) NumPDs: 1 Channels: 1
OSDP: CP: PD-0: [2024-02-11T19:05:05] [ERROR] src/osdp_cp.c:666: Failed to verify PD cryptogram
OSDP: CP: PD-0: [2024-02-11T19:05:05] [WARN ] src/osdp_cp.c:1077: SC Failed. Retry with SCBK-D
OSDP: CP: PD-0: [2024-02-11T19:05:05] [WARN ] src/osdp_cp.c:1040: SC ACtive with SCBK-D. Set SCBK
OSDP: CP: PD-0: [2024-02-11T19:05:05] [ERROR] src/osdp_cp.c:331: Cannot perform KEYSET without SC!
OSDP: CP: PD-0: [2024-02-11T19:05:05] [ERROR] src/osdp_cp.c:853: Failed to build packet for CMD(117)
OSDP: CP: PD-0: [2024-02-11T19:05:05] [ERROR] src/osdp_cp.c:1092: Failed to set SCBK; Set PD offline due to ENFORCE_SECURE
OSDP: CP: PD-0: [2024-02-11T19:05:05] [ERROR] src/osdp_cp.c:1172: Going offline for 1 seconds; Was in 'SC-SetSCBK' state
OSDP: CP: PD-0: [2024-02-11T19:05:06] [ERROR] src/osdp_cp.c:666: Failed to verify PD cryptogram
OSDP: CP: PD-0: [2024-02-11T19:05:06] [WARN ] src/osdp_cp.c:1077: SC Failed. Retry with SCBK-D
OSDP: CP: PD-0: [2024-02-11T19:05:07] [WARN ] src/osdp_cp.c:1040: SC ACtive with SCBK-D. Set SCBK
OSDP: CP: PD-0: [2024-02-11T19:05:07] [ERROR] src/osdp_cp.c:331: Cannot perform KEYSET without SC!
OSDP: CP: PD-0: [2024-02-11T19:05:07] [ERROR] src/osdp_cp.c:853: Failed to build packet for CMD(117)
OSDP: CP: PD-0: [2024-02-11T19:05:07] [ERROR] src/osdp_cp.c:1092: Failed to set SCBK; Set PD offline due to ENFORCE_SECURE
OSDP: CP: PD-0: [2024-02-11T19:05:07] [ERROR] src/osdp_cp.c:1172: Going offline for 2 seconds; Was in 'SC-SetSCBK' state

Example program to reproduce:

#!/usr/bin/python

import time
import sys
from osdp import *

key = KeyStore.gen_key()

pd_info = [
  PDInfo(00, scbk=key, name='door-00')
]

pd_info[0].channel_type='uart';
pd_info[0].channel_speed=115200;
pd_info[0].channel_device='/dev/ttyUSB0';

cp = ControlPanel(pd_info, log_level=LogLevel.Debug)

cp.start()
cp.sc_wait_all()

sidcha, 2024-02-11:
Ah, sorry. Found the issue.. it is a regression from the latest CP state machine clean up. Fixing it now.

closed 2024-02-11 by sidcha

morcant, 2024-02-12:
Confirmed, thanks!

commit 13b39849 (2024-02-11): libosdp: cp: Fix broken keyset for an install_mode PD

When the PD is in install mode, the CP should be able to establish a
secure channel with the default SCBK and then use that sesssion to set a
new key (the one the CP recieved for that PD during setup).

This used to work before but was broken during the CP state machine
cleanup done in b4cfb576ab and missing tests for this case allowed that
regression to go unnoticed (this issue was addressed in the last
commit).

This commit fixed the regression.

Fixes: #157
Fixes: b4cfb576ab603fdd434a1883d475ac00ab84e61c