osdp-dev/libosdp #159: Unable to install via Pypi

denravonska, 2024-02-13:
libosdp is available as a package on [Pypi](https://pypi.org/project/libosdp/). According to the readme it can be installed using `pip install libosdp`, but that fails with

```bash
(env) marco@dev:~/temp$ pip install libosdp
Collecting libosdp
  Downloading libosdp-2.4.0.tar.gz (8.2 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [19 lines of output]
      Traceback (most recent call last):
        File "/home/marco/temp/env/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/home/marco/temp/env/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/home/marco/temp/env/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-f1uj2zdn/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 325, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=['wheel'])
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-f1uj2zdn/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 295, in _get_build_requires
          self.run_setup()
        File "/tmp/pip-build-env-f1uj2zdn/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 311, in run_setup
          exec(code, locals())
        File "<string>", line 25, in <module>
        File "<string>", line 18, in read_version
      FileNotFoundError: [Errno 2] No such file or directory: '/home/marco/CMakeLists.txt'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

It seems to assume a very particular file structure on the host that runs the command. When building from source this makes sense as it's the source tree structure, but outside of that it becomes very difficult to install.

denravonska, 2024-02-13:
Installing from source via git+https works perfectly. Thanks again :)

sidcha, 2024-02-14:
With the latest release, I can install from PyPI without any issues.
```
[ sidcha@Sid-MacBook-Pro: rust (master) ]$ python3 -m venv env
[ sidcha@Sid-MacBook-Pro: rust (master) ]$ . ./env/bin/activate
(env) [ sidcha@Sid-MacBook-Pro: rust (master) ]$ pip install libosdp
Collecting libosdp
  Downloading libosdp-3.0.0-cp311-cp311-macosx_11_0_arm64.whl.metadata (2.9 kB)
Downloading libosdp-3.0.0-cp311-cp311-macosx_11_0_arm64.whl (69 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 69.2/69.2 kB 2.4 MB/s eta 0:00:00
Installing collected packages: libosdp
Successfully installed libosdp-3.0.0

[notice] A new release of pip is available: 23.3.1 -> 24.0
[notice] To update, run: pip install --upgrade pip
(env) [ sidcha@Sid-MacBook-Pro: rust (master) ]$ python3
Python 3.11.6 (main, Oct  2 2023, 13:45:54) [Clang 15.0.0 (clang-1500.0.40.1)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import osdp
>>> exit()
```
Closing this issue as complete.

closed 2024-02-14 by sidcha