osdp-dev/libosdp #166: CP send a broadcast

jason155392, 2024-02-21:
Can CP send a broadcast to PD now?
I checked the API and found that only the osdp_cp_send_command function is available.
This function requires specifying the PD offset.
However, osdp_cp_setup cannot be set up with osdp_pd_info_t for the 0x7F address.
This will cause a "packet seq mismatch" issue.

How can libosdp handle this now?

sidcha, 2024-02-21:
The broadcast support that we have is entirely for the PD so if you need it for the CP, we should enable it. Can you describe what you are trying to achieve?

Also, as I recall, the broadcast message was underspecified in OSDP v2.2 for all practical purposes.

jason155392, 2024-02-22:
I want to implement a function to get/set a Peripheral Device's (PD) address using broadcast messages.

Initially, I'm considering the following approach:
1. The Control Panel (CP) can send an "osdp_MFG" broadcast to check the PD's address.
2. Subsequently, the CP can send an "osdp_COMSET" broadcast to set the PD's address.

I've reviewed OSDP v2.2, which specifies that "the use of the broadcast address should be limited to controlled (single PD) configurations."
Therefore, this method is suitable for closed environments, where the CP can utilize it to manage single PD information effectively.

sidcha, 2024-02-22:
Okay, that sounds compelling enough. Let me see how this can be implemented.

sidcha, 2024-02-22:
Looks like #113 can make use of this too.

schmida2, 2024-03-13:
> Looks like #113 can make use of this too.

The proposed concept of OSDP auto addressing revision 2 does not rely on broadcast anymore, esp. because using broadcasts during operation sets back message sequence numbers (SQN) and this would harm the OSDP built-in error recovery

sidcha, 2024-03-13:
@schmida2 I turned #113 into a [discussion](https://github.com/goToMain/libosdp/discussions/171). Do you mind updating that thread with the most recent version of your proposal?

(I think the current link points to the one with broadcast messages)

dhopsonlb, 2024-07-12:
Is there any news on this functionality? We're using libosdp professionally and CP broadcast support is important for us.

sidcha, 2024-07-20:
@dhopsonlb, What use case are you targeting? Is it the same as @jason155392? If not, the solution I had in mind (relatively simpler approach) would not work for the general case "CP can send any command as broadcast".

commit 2c825cb2 (2025-03-02): API: CP: Add support for command broadcast

LibOSDP supports broadcast packets for PD but does not have any support
for broadcast of CP app commands. This patch adds support for CP to send
certain commands as broadcast packets to the PD.

The following restrictions apply:
- Only single PD environments will accept commands with broacast flag
- When ENFORCE_SECURE is enabled, commands with broacast flags are
  rejected.

Fixes: #166
closed 2025-03-02 by sidcha