osdp-dev/libosdp #181: Request for Help: How to Query PD for Firmware Version

boobika, 2024-05-28:
Hello,

I am using the goToMain/libosdp library in Python and I am trying to query a Peripheral Device (PD) to determine its firmware version. According to the OSDP protocol documentation, this should be possible using a command identified by CMD_ID, as mentioned here: [OSDP Commands and Replies](https://libosdp.sidcha.dev/protocol/commands-and-replies.html).

However, I couldn't find an appropriate command within the available commands such as OSDP_CMD_LED, OSDP_CMD_BUZZER, etc., to retrieve the firmware version.

Could someone please guide me on how to query the PD for its firmware version using goToMain/libosdp?

Thank you in advance for your assistance!

Best regards,
Bob

sidcha, 2024-05-28:
LibOSDP fetches this information during handshake and application can get this info using the C API `osdp_cp_get_pd_id()`. It is currently not exposed to the python layer.

Will enable this later today. Thanks for reporting the issue.

sidcha, 2024-05-28:
Now you can use,

```
cp.get_pd_id(pd.address)
```

and 

```
cp.check_capability(pd.address, Capability.CommunicationSecurity)
```
to access the PD_ID information and Capabilities as reported by the PD to LibOSDP.

closed 2024-05-28 by sidcha

boobika, 2024-05-29:
Thank you for adding the functionality I requested so quickly, especially on the same day. I’m going to test it out now.

boobika, 2024-05-29:
Hello,

I am trying **cp.get_pd_id(pd.address)** and it returns the values as seen in the log below. The value for firmware_version looks like the maximum value of some type in the C language?
The reader is **Secustos SQ80 LEGIC** from the manufacturer sesamsec.

```
pyosdp: [2024-05-29T12:45:51Z] [INFO ] vendor/src/osdp_cp.c:1372: Setup complete; LibOSDP-3.0.6 master (c95b95f) NumPDs: 1 Channels: 1.
OSDP: CP: PD-1: [2024-05-29T12:45:51Z] [DEBUG] vendor/src/osdp_cp.c:684: CMD: ID(61) REPLY: PDID(45)
OSDP: CP: PD-1: [2024-05-29T12:45:51Z] [DEBUG] vendor/src/osdp_cp.c:1185: StateChange: [ID-Request] -> [Cap-Detect] (SC-Inactive)
OSDP: CP: PD-1: [2024-05-29T12:45:51Z] [DEBUG] vendor/src/osdp_cp.c:460: Reports capability 'ContactStatusMonitoring' (1/0)
OSDP: CP: PD-1: [2024-05-29T12:45:51Z] [DEBUG] vendor/src/osdp_cp.c:460: Reports capability 'OutputControl' (1/0)
OSDP: CP: PD-1: [2024-05-29T12:45:51Z] [DEBUG] vendor/src/osdp_cp.c:460: Reports capability 'CardDataFormat' (3/0)
OSDP: CP: PD-1: [2024-05-29T12:45:51Z] [DEBUG] vendor/src/osdp_cp.c:460: Reports capability 'LEDControl' (4/1)
OSDP: CP: PD-1: [2024-05-29T12:45:51Z] [DEBUG] vendor/src/osdp_cp.c:460: Reports capability 'AudibleControl' (2/1)
OSDP: CP: PD-1: [2024-05-29T12:45:51Z] [DEBUG] vendor/src/osdp_cp.c:460: Reports capability 'TextOutput' (0/0)
OSDP: CP: PD-1: [2024-05-29T12:45:51Z] [DEBUG] vendor/src/osdp_cp.c:460: Reports capability 'TimeKeeping' (0/0)
OSDP: CP: PD-1: [2024-05-29T12:45:51Z] [DEBUG] vendor/src/osdp_cp.c:460: Reports capability 'CheckCharacter' (1/0)
OSDP: CP: PD-1: [2024-05-29T12:45:51Z] [DEBUG] vendor/src/osdp_cp.c:460: Reports capability 'CommunicationSecurity' (1/1)
OSDP: CP: PD-1: [2024-05-29T12:45:51Z] [DEBUG] vendor/src/osdp_cp.c:460: Reports capability 'ReceiveBufferSize' (149/0)
OSDP: CP: PD-1: [2024-05-29T12:45:51Z] [DEBUG] vendor/src/osdp_cp.c:460: Reports capability 'CombinedMessageSize' (149/0)
OSDP: CP: PD-1: [2024-05-29T12:45:51Z] [DEBUG] vendor/src/osdp_cp.c:460: Reports capability 'SmartCard' (0/0)
OSDP: CP: PD-1: [2024-05-29T12:45:51Z] [DEBUG] vendor/src/osdp_cp.c:460: Reports capability 'Reader' (0/1)
OSDP: CP: PD-1: [2024-05-29T12:45:51Z] [DEBUG] vendor/src/osdp_cp.c:460: Reports capability 'Biometric' (0/0)
OSDP: CP: PD-1: [2024-05-29T12:45:51Z] [DEBUG] vendor/src/osdp_cp.c:684: CMD: CAP(62) REPLY: PDCAP(46)
OSDP: CP: PD-1: [2024-05-29T12:45:51Z] [DEBUG] vendor/src/osdp_cp.c:1185: StateChange: [Cap-Detect] -> [SC-Chlng] (SC-Inactive)
OSDP: CP: PD-1: [2024-05-29T12:45:51Z] [DEBUG] vendor/src/osdp_cp.c:684: CMD: CHLNG(76) REPLY: CCRYPT(76)
OSDP: CP: PD-1: [2024-05-29T12:45:51Z] [DEBUG] vendor/src/osdp_cp.c:1185: StateChange: [SC-Chlng] -> [SC-Scrypt] (SC-Inactive)
OSDP: CP: PD-1: [2024-05-29T12:45:51Z] [DEBUG] vendor/src/osdp_cp.c:684: CMD: SCRYPT(77) REPLY: RMAC_I(78)
OSDP: CP: PD-1: [2024-05-29T12:45:51Z] [INFO ] vendor/src/osdp_cp.c:1162: Online; With SC
OSDP: CP: PD-1: [2024-05-29T12:45:51Z] [DEBUG] vendor/src/osdp_cp.c:1185: StateChange: [SC-Scrypt] -> [Online] (SC-Active)
App: [2024-05-29 14:45:52] [INFO] version: 0
App: [2024-05-29 14:45:52] [INFO] model: 3
App: [2024-05-29 14:45:52] [INFO] vendor_code: 204064
App: [2024-05-29 14:45:52] [INFO] serial_number: 182882549
App: [2024-05-29 14:45:52] [INFO] firmware_version: 65536
```

sidcha, 2024-05-29:
Firmware version is 3 bytes (major, minor, and build) which is then packed into a int by LibOSDP. In this case the reader is reporting 1.0.0 as firmware_version (65536 is 0x10000 so [0x01, 0x00, 0x00]).

I could split this python wrapper.

boobika, 2024-05-29:
Ah, I apologize. In this case, it is correct.