osdp-dev/libosdp #191: FileTransfer finishing incorrectly with last FILETRANSFER repeated again

xsrf, 2024-08-14:
I made a python script to perform a firmware update based on the code from the screenshot in https://github.com/goToMain/libosdp/issues/185#issuecomment-2195026272

The update works, but I noticed that the library is not ending the transfer correctly.

The PD I have responds to the last packet with `FTSTAT` and `FtStatusDetail = 2` (reboot, expect reset). 
But then the CP sends the last packet again, which the PD responds to with NACK (9) and the CP tears down the connection.

```
OSDP: CP: PD-1: [2024-08-14T11:14:59Z] [DEBUG] vendor\src/osdp_cp.c:686: CMD: FILETRANSFER(7c) REPLY: FTSTAT(7a)
OSDP: CP: PD-1: [2024-08-14T11:14:59Z] [DEBUG] vendor\src/osdp_cp.c:686: CMD: FILETRANSFER(7c) REPLY: FTSTAT(7a)
OSDP: CP: PD-1: [2024-08-14T11:15:00Z] [DEBUG] vendor\src/osdp_cp.c:686: CMD: FILETRANSFER(7c) REPLY: FTSTAT(7a)
OSDP: CP: PD-1: [2024-08-14T11:15:00Z] [WARN ] vendor\src/osdp_cp.c:423: PD replied with NAK(9) for CMD(7c)
OSDP: CP: PD-1: [2024-08-14T11:15:00Z] [DEBUG] vendor\src/osdp_cp.c:686: CMD: FILETRANSFER(7c) REPLY: NAK(41)
OSDP: CP: PD-1: [2024-08-14T11:15:00Z] [ERROR] vendor\src/osdp_cp.c:1177: Going offline for 1 seconds; Was in 'Online' state
```

I captured the communication using Wireshark, since I do use a Ethernet to RS485 converter:
```
No.     Time           Source                Destination           Protocol Length Info
  62223 2725.442374    192.168.3.211         192.168.3.145         OSDP     190    Plaintext Message (FILETRANSFER)

Transmission Control Protocol, Src Port: 62428, Dst Port: 4196, Seq: 332210, Ack: 39223, Len: 136
OSDP Packet
    Header: ff5301870005
    Payload: 7c0150530400885204007400000000000000000000000000000000000000000000000000…
        Command/Reply ID: 0x7c
        Command/Reply Data: 015053040088520400740000000000000000000000000000000000000000000000000000…
    CheckSum: 91cc

No.     Time           Source                Destination           Protocol Length Info
  62225 2725.607617    192.168.3.145         192.168.3.211         OSDP     70     Plaintext Message (FTSTAT)

Transmission Control Protocol, Src Port: 4196, Dst Port: 62428, Seq: 39223, Ack: 332346, Len: 16
OSDP Packet
    Header: ff53810f0005
    Payload: 7a00000000000000
        Command/Reply ID: 0x7a
        Command/Reply Data: 00000000000000
    CheckSum: 59b9

No.     Time           Source                Destination           Protocol Length Info
  62226 2725.650009    192.168.3.211         192.168.3.145         OSDP     158    Plaintext Message (FILETRANSFER)

Transmission Control Protocol, Src Port: 62428, Dst Port: 4196, Seq: 332346, Ack: 39239, Len: 104
OSDP Packet
    Header: ff5301670006
    Payload: 7c0150530400fc5204005400c71f002057ffffffff800000000000000000000000000000…
        Command/Reply ID: 0x7c
        Command/Reply Data: 0150530400fc5204005400c71f002057ffffffff80000000000000000000000000000000…
    CheckSum: 01c0

No.     Time           Source                Destination           Protocol Length Info
  62228 2725.784576    192.168.3.145         192.168.3.211         OSDP     70     Plaintext Message (FTSTAT)

Transmission Control Protocol, Src Port: 4196, Dst Port: 62428, Seq: 39239, Ack: 332450, Len: 16
OSDP Packet
    Header: ff53810f0006
    Payload: 7a00000002000000
        Command/Reply ID: 0x7a
        Command/Reply Data: 00000002000000
    CheckSum: 7579

No.     Time           Source                Destination           Protocol Length Info
  62230 2725.826031    192.168.3.211         192.168.3.145         OSDP     158    Plaintext Message (FILETRANSFER)

Transmission Control Protocol, Src Port: 62428, Dst Port: 4196, Seq: 332450, Ack: 39255, Len: 104
OSDP Packet
    Header: ff5301670007
    Payload: 7c0150530400fc5204005400c71f002057ffffffff800000000000000000000000000000…
        Command/Reply ID: 0x7c
        Command/Reply Data: 0150530400fc5204005400c71f002057ffffffff80000000000000000000000000000000…
    CheckSum: 3b5e

No.     Time           Source                Destination           Protocol Length Info
  62232 2725.950622    192.168.3.145         192.168.3.211         OSDP     64     Plaintext Message (NACK)

Transmission Control Protocol, Src Port: 4196, Dst Port: 62428, Seq: 39255, Ack: 332554, Len: 10
OSDP Packet
    Header: ff5381090007
    Payload: 4109
        Command/Reply ID: 0x41
        Command/Reply Data: 09
    CheckSum: df39
```

You can see that the last two `FILETRANSFER` are shorter (as it's the end - I confirmed this looking at the data) and have identical payload. You can also see the last `FTSTAT` with detail `2`.

I guess this is an issue with the library and nothing I could have messed up?

This is the script I'm using:
```python
#!/usr/bin/env python3

import argparse
import socket
import time
import os

from osdp import *

class TCPChannel:
    def __init__(self, host: str, port: int):
        self.sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
        self.sock.connect((host, port))

    def read(self, max_read: int):
        return self.sock.recv(max_read)

    def write(self, data: bytes):
        return self.sock.send(data)

    def flush(self):
        self.sock.setsockopt(socket.SOL_SOCKET, socket.SO_SNDBUF, 1)

    def __del__(self):
        self.sock.close()

parser = argparse.ArgumentParser(prog = 'osdp_fw_update.py', description = "LibOSDP CP APP")
parser.add_argument("device", type = str, metavar = "IPADDR", help = "IP-Address or hostname of Reader/RS485 Converter Port 4196")
parser.add_argument("reader", type = int, metavar = "READER", help = "Reader-ID on Bus (default: 1)")
parser.add_argument("filename", type = str, metavar = "FILENAME", help = "Filename *.osdp")
parser.add_argument("--port", type = int, metavar = "PORT", default = 4196, help = "Reader/RS485 Converter Port (default: 4196)")
parser.add_argument("--loglevel", type = int, metavar = "LEVEL", default = 6, help = "LibOSDP log level; can be 0-7 (default: 6)")
args = parser.parse_args()

sender_data = None
sender_delay = 1

def file_read(filename):
    global sender_data
    with open(filename,"rb") as file:
        sender_data = file.read()
        print(f"File read {filename}")

def sender_open(file_id: int, file_size: int) -> int:
    global sender_data
    return len(sender_data)

def sender_read(size: int, offset: int) -> bytes:
    if sender_delay > 0:
        time.sleep(sender_delay)
    global sender_data
    sender_data_length = len(sender_data)
    if offset + size > sender_data_length:
        size = sender_data_length - offset
    return bytes(sender_data[offset:offset+size])

def sender_write(data: bytes, offset: int) -> int:
    return False

def sender_close(file_id: int):
    return True

sender_fops = {
    'open': sender_open,
    'read': sender_read,
    'write': sender_write,
    'close': sender_close
}

def send_file_cmd(cp: ControlPanel, pd_info_address: int, file_id: int):
    file_read(args.filename)
    result = cp.register_file_ops(pd_info_address,sender_fops)
    print(f"Register file ops {result} {pd_info_address} {file_id}")
    file_tx_cmd = {
        'command': Command.FileTransfer,
        'id': file_id,
        'flags': 0
    }
    result = cp.send_command(pd_info_address, file_tx_cmd)
    if result:
        print("FW Upgrade started... Reader will reboot when done!")
    else:
        print("FW Upgrade ERROR")

try:
    ## Describe the PD (setting scbk=None puts the PD in install mode)
    channel = TCPChannel(args.device, args.port)
    pd_info = [
        PDInfo(args.reader, channel, scbk=0),
    ]

    ## Create a CP device and kick-off the handler thread
    cp = ControlPanel(pd_info, log_level=args.loglevel)
    print('cp.start()')
    cp.start()
    cp.online_wait_all()

    beep_cmd = {
        'command': Command.Buzzer,
        'reader': 0,
        'control_code': 2,
        'on_count': 1,
        'off_count': 1,
        'rep_count': 1
    }

    print('Loop')
    cp.send_command(pd_info[0].address, beep_cmd)
    time.sleep(0.5)
    send_file_cmd(cp, pd_info[0].address, 1)
    time.sleep(2)
    sender_delay = 0
    while True:
        event = cp.get_event(pd_info[0].address, timeout=2)
        status = cp.get_file_tx_status(pd_info[0].address)
        pct = round((status['offset']/status['size'])*100 , 1)
        print(f"Upload: {status['offset']} / {status['size']} ({pct}%)", end="\r", flush=True)

except KeyboardInterrupt:
    print("Quit")
    os._exit(0)

except:
    print("Quit")
    os._exit(0)
```

sidcha, 2024-08-15:
@xsrf Thanks for the details bug report. Please check if this works for you.

xsrf, 2024-08-16:
Well I don't get the last `FILETRANSFER` repeated anymore, but now the CP suddenly wants to init a SC:
```
OSDP: CP: PD-1: 2024-08-16T17:18:59Z src/osdp_cp.c:682  [DEBUG] CMD: FILETRANSFER(7c) REPLY: FTSTAT(7a)
OSDP: CP: PD-1: 2024-08-16T17:18:59Z src/osdp_cp.c:682  [DEBUG] CMD: FILETRANSFER(7c) REPLY: FTSTAT(7a)
OSDP: CP: PD-1: 2024-08-16T17:19:00Z src/osdp_cp.c:682  [DEBUG] CMD: FILETRANSFER(7c) REPLY: FTSTAT(7a)
OSDP: CP: PD-1: 2024-08-16T17:19:00Z src/osdp_cp.c:682  [DEBUG] CMD: FILETRANSFER(7c) REPLY: FTSTAT(7a)
OSDP: CP: PD-1: 2024-08-16T17:19:00Z src/osdp_cp.c:682  [DEBUG] CMD: FILETRANSFER(7c) REPLY: FTSTAT(7a)
OSDP: CP: PD-1: 2024-08-16T17:19:00Z src/osdp_cp.c:682  [DEBUG] CMD: FILETRANSFER(7c) REPLY: FTSTAT(7a)
OSDP: CP: PD-1: 2024-08-16T17:19:00Z src/osdp_cp.c:682  [DEBUG] CMD: FILETRANSFER(7c) REPLY: FTSTAT(7a)
OSDP: CP: PD-1: 2024-08-16T17:19:01Z src/osdp_file.c:184  [INFO ] Stat_Decode: File transfer complete
OSDP: CP: PD-1: 2024-08-16T17:19:01Z src/osdp_cp.c:682  [DEBUG] CMD: FILETRANSFER(7c) REPLY: FTSTAT(7a)
OSDP: CP: PD-1: 2024-08-16T17:19:01Z src/osdp_cp.c:1229 [DEBUG] StateChange: [Online] -> [SC-Chlng] (SC-Inactive)
OSDP: CP: PD-1: 2024-08-16T17:19:01Z src/osdp_cp.c:419  [WARN ] PD replied with NAK(5) for CMD: CHLNG(76)
OSDP: CP: PD-1: 2024-08-16T17:19:01Z src/osdp_cp.c:682  [DEBUG] CMD: CHLNG(76) REPLY: NAK(41)
OSDP: CP: PD-1: 2024-08-16T17:19:01Z src/osdp_cp.c:1156 [WARN ] SC Failed. Retry with SCBK-D
OSDP: CP: PD-1: 2024-08-16T17:19:01Z src/osdp_cp.c:419  [WARN ] PD replied with NAK(5) for CMD: CHLNG(76)
OSDP: CP: PD-1: 2024-08-16T17:19:01Z src/osdp_cp.c:682  [DEBUG] CMD: CHLNG(76) REPLY: NAK(41)
OSDP: CP: PD-1: 2024-08-16T17:19:01Z src/osdp_cp.c:1205 [INFO ] Online; Without SC
OSDP: CP: PD-1: 2024-08-16T17:19:01Z src/osdp_cp.c:1229 [DEBUG] StateChange: [SC-Chlng] -> [Online] (SC-Inactive)
Quit (me pressing STRG-C because it hang)
(had to kill python via Task-Manager)
```

The capture flag didn't work (I added `cp.teardown()` to my exception handler which caused it to freeze) so here is my ethernet capture again, which should be enough anyways:
```
No.     Time           Source                Destination           Protocol Length Info
  17239 621.184695     192.168.3.211         192.168.3.145         OSDP     150    Plaintext Message (FILETRANSFER)

Transmission Control Protocol, Src Port: 51909, Dst Port: 4196, Seq: 332346, Ack: 39239, Len: 96
OSDP Packet
    Header: ff53015f0006
    Payload: 7c0148530400fc5204004c…
        Command/Reply ID: 0x7c
        Command/Reply Data: 0148530400fc5204004c…
    CheckSum: af4d

No.     Time           Source                Destination           Protocol Length Info
  17244 621.310571     192.168.3.145         192.168.3.211         OSDP     70     Plaintext Message (FTSTAT)

Transmission Control Protocol, Src Port: 4196, Dst Port: 51909, Seq: 39239, Ack: 332442, Len: 16
OSDP Packet
    Header: ff53810f0006
    Payload: 7a00000002000000
        Command/Reply ID: 0x7a
        Command/Reply Data: 00000002000000
    CheckSum: 7579

No.     Time           Source                Destination           Protocol Length Info
  17247 621.352175     192.168.3.211         192.168.3.145         OSDP     74     Seure Message

Transmission Control Protocol, Src Port: 51909, Dst Port: 4196, Seq: 332442, Ack: 39255, Len: 20
OSDP Packet
    Header: ff530113000c
    Payload: 03110176009031cf957a59e5
        Command/Reply ID: 0x76
        Secure Channel Block: 031101 (SCS_11)
    CheckSum: 9325

No.     Time           Source                Destination           Protocol Length Info
  17252 621.388600     192.168.3.145         192.168.3.211         OSDP     64     Plaintext Message (NACK)

Transmission Control Protocol, Src Port: 4196, Dst Port: 51909, Seq: 39255, Ack: 332462, Len: 10
OSDP Packet
    Header: ff5381090004
    Payload: 4105
        Command/Reply ID: 0x41
        Command/Reply Data: 05
    CheckSum: 03a1

No.     Time           Source                Destination           Protocol Length Info
[... 75 lines cut ...]

No.     Time           Source                Destination           Protocol Length Info
  17268 621.795510     192.168.3.145         192.168.3.211         OSDP     63     Plaintext Message (ACK)

Transmission Control Protocol, Src Port: 4196, Dst Port: 51909, Seq: 39293, Ack: 332509, Len: 9
OSDP Packet
    Header: ff5381080005
    Payload: 40
        Command/Reply ID: 0x40
        Command/Reply Data: <MISSING>
    CheckSum: 3935

No.     Time           Source                Destination           Protocol Length Info
  17270 621.877653     192.168.3.211         192.168.3.145         OSDP     63     Plaintext Message (POLL)

Transmission Control Protocol, Src Port: 51909, Dst Port: 4196, Seq: 332509, Ack: 39302, Len: 9
OSDP Packet
    Header: ff5301080006
    Payload: 60
        Command/Reply ID: 0x60
        Command/Reply Data: <MISSING>
    CheckSum: d866

*** reader reboot ***
```

sidcha, 2024-08-16:
I think(I cannot say for sure because data potion is missing) the PD responded with “File contents processed” instead of “going to reboot”. In this case, CP just goes on about its normal tasks - one if which is to attempt a secure channel if the PD is capable of doing so.

There is this other mode where the PD can ask the file transfer to be in plain text which LibOSDP ignores on purpose because I think it could have security implications.

xsrf, 2024-08-16:
> I think(I cannot say for sure because data potion is missing) the PD responded with “File contents processed” instead of “going to reboot”.

```
No.     Time           Source                Destination           Protocol Length Info
  17244 621.310571     192.168.3.145         192.168.3.211         OSDP     70     Plaintext Message (FTSTAT)

Transmission Control Protocol, Src Port: 4196, Dst Port: 51909, Seq: 39239, Ack: 332442, Len: 16
OSDP Packet
    Header: ff53810f0006
    Payload: 7a00000002000000
        Command/Reply ID: 0x7a
        Command/Reply Data: 00 00 00 02 00 00 00
    CheckSum: 7579
```

Payload is not missing, it's 0x02 "rebooting now, expect full communications reset."

Also, PD is not capable of "CommunicationSecurity" if I read this correctly
```
>python osdp_fw_update.py 192.168.3.145 1 sq80.osdp --loglevel=7
pyosdp: 2024-08-16T19:45:18Z src/osdp_cp.c:1462 [INFO ] CP Setup complete; LibOSDP-3.0.6 master (99ab23f) NumPDs:1 Channels:1
cp.start()
OSDP: CP: PD-1: 2024-08-16T19:45:18Z src/osdp_cp.c:682  [DEBUG] CMD: ID(61) REPLY: PDID(45)
OSDP: CP: PD-1: 2024-08-16T19:45:18Z src/osdp_cp.c:1229 [DEBUG] StateChange: [ID-Request] -> [Cap-Detect] (SC-Inactive)
OSDP: CP: PD-1: 2024-08-16T19:45:18Z src/osdp_cp.c:459  [DEBUG] Reports capability 'ContactStatusMonitoring' (1/0)
OSDP: CP: PD-1: 2024-08-16T19:45:18Z src/osdp_cp.c:459  [DEBUG] Reports capability 'OutputControl' (1/0)
OSDP: CP: PD-1: 2024-08-16T19:45:18Z src/osdp_cp.c:459  [DEBUG] Reports capability 'CardDataFormat' (3/0)
OSDP: CP: PD-1: 2024-08-16T19:45:18Z src/osdp_cp.c:459  [DEBUG] Reports capability 'LEDControl' (4/2)
OSDP: CP: PD-1: 2024-08-16T19:45:18Z src/osdp_cp.c:459  [DEBUG] Reports capability 'AudibleControl' (2/1)
OSDP: CP: PD-1: 2024-08-16T19:45:18Z src/osdp_cp.c:459  [DEBUG] Reports capability 'TextOutput' (0/0)
OSDP: CP: PD-1: 2024-08-16T19:45:18Z src/osdp_cp.c:459  [DEBUG] Reports capability 'TimeKeeping' (0/0)
OSDP: CP: PD-1: 2024-08-16T19:45:18Z src/osdp_cp.c:459  [DEBUG] Reports capability 'CheckCharacter' (1/0)
OSDP: CP: PD-1: 2024-08-16T19:45:18Z src/osdp_cp.c:459  [DEBUG] Reports capability 'CommunicationSecurity' (0/0)
OSDP: CP: PD-1: 2024-08-16T19:45:18Z src/osdp_cp.c:459  [DEBUG] Reports capability 'ReceiveBufferSize' (149/0)
OSDP: CP: PD-1: 2024-08-16T19:45:18Z src/osdp_cp.c:459  [DEBUG] Reports capability 'CombinedMessageSize' (149/0)
OSDP: CP: PD-1: 2024-08-16T19:45:18Z src/osdp_cp.c:459  [DEBUG] Reports capability 'SmartCard' (0/0)
OSDP: CP: PD-1: 2024-08-16T19:45:18Z src/osdp_cp.c:459  [DEBUG] Reports capability 'Reader' (0/1)
OSDP: CP: PD-1: 2024-08-16T19:45:18Z src/osdp_cp.c:459  [DEBUG] Reports capability 'Biometric' (0/0)
OSDP: CP: PD-1: 2024-08-16T19:45:18Z src/osdp_cp.c:682  [DEBUG] CMD: CAP(62) REPLY: PDCAP(46)
OSDP: CP: PD-1: 2024-08-16T19:45:18Z src/osdp_cp.c:1205 [INFO ] Online; Without SC
OSDP: CP: PD-1: 2024-08-16T19:45:18Z src/osdp_cp.c:1229 [DEBUG] StateChange: [Cap-Detect] -> [Online] (SC-Inactive)
```

And I initialized scbk with 0 (not sure how to disable SC on CP side properly)
```
PDInfo(args.reader, channel, scbk=0, flags=[LibFlag.CapturePackets]),
```

sidcha, 2024-08-17:
> Payload is not missing, it's 0x02 "rebooting now, expect full communications reset."

Right. I was replying from phone with text wraps :) There were some `<MISSING>` tags so I assumed the payload part is just missing everywhere. I believe now this issue must be fixed.

> And I initialized scbk with 0 (not sure how to disable SC on CP side properly)

Setting scbk to anything other than `Bytes` type will cause CP to disable SC for that PD so this is fine (although it would be more idiomatic to set it to `None`). As to why CP attempted to start a secure channel in spite this was a regression - more info in commit message of [485870b](https://github.com/goToMain/libosdp/commit/485870b3826ea9cffcd293004beefa216508181f) if you are curious why.

xsrf, 2024-08-17:
Great, works. It now goes offline immediately after the last `FTSTAT`, waits 5min and re-inits correctly again.
Is the 5min configurable?
```
...
OSDP: CP: PD-1: 2024-08-17T09:26:48Z src/osdp_cp.c:682  [DEBUG] CMD: FILETRANSFER(7c) REPLY: FTSTAT(7a)
OSDP: CP: PD-1: 2024-08-17T09:26:48Z src/osdp_cp.c:682  [DEBUG] CMD: FILETRANSFER(7c) REPLY: FTSTAT(7a)
OSDP: CP: PD-1: 2024-08-17T09:26:48Z src/osdp_cp.c:682  [DEBUG] CMD: FILETRANSFER(7c) REPLY: FTSTAT(7a)
OSDP: CP: PD-1: 2024-08-17T09:26:48Z src/osdp_cp.c:682  [DEBUG] CMD: FILETRANSFER(7c) REPLY: FTSTAT(7a)
OSDP: CP: PD-1: 2024-08-17T09:26:48Z src/osdp_cp.c:682  [DEBUG] CMD: FILETRANSFER(7c) REPLY: FTSTAT(7a)
OSDP: CP: PD-1: 2024-08-17T09:26:49Z src/osdp_file.c:184  [INFO ] Stat_Decode: File transfer complete
OSDP: CP: PD-1: 2024-08-17T09:26:49Z src/osdp_cp.c:682  [DEBUG] CMD: FILETRANSFER(7c) REPLY: FTSTAT(7a)
OSDP: CP: PD-1: 2024-08-17T09:26:49Z src/osdp_cp.c:1317 [INFO ] Going offline due to request
OSDP: CP: PD-1: 2024-08-17T09:26:49Z src/osdp_cp.c:1214 [ERROR] Going offline for 300 seconds; Was in 'Online' state
OSDP: CP: PD-1: 2024-08-17T09:26:49Z src/osdp_cp.c:1229 [DEBUG] StateChange: [Online] -> [Offline] (SC-Inactive)
OSDP: CP: PD-1: 2024-08-17T09:31:49Z src/osdp_cp.c:1229 [DEBUG] StateChange: [Offline] -> [ID-Request] (SC-Inactive)
OSDP: CP: PD-1: 2024-08-17T09:31:49Z src/osdp_cp.c:682  [DEBUG] CMD: ID(61) REPLY: PDID(45)
OSDP: CP: PD-1: 2024-08-17T09:31:49Z src/osdp_cp.c:1229 [DEBUG] StateChange: [ID-Request] -> [Cap-Detect] (SC-Inactive)
...
```
`LibOSDP-3.0.6 master (485870b)`

sidcha, 2024-08-17:
That value (and other timeouts) comes from src/config.h.in and changes require the library to be rebuilt to take effect.

closed 2024-08-17 by sidcha