osdp-dev/libosdp #194: Problem with "Discard Secure Channel if a KEYSET is ACKed in plaintext" in PD mode

ldholmgren, 2024-08-26:
**Problem**
We are working on implementing libosdp in our next generation of products with the goal of replacing the current 8+ year old proprietary OSDP library we have in production. We need it to be 100% compatible with our existing products in the field, as well as with our partners, several of whom use the same proprietary OSDP library.

We have hit somewhat of a roadblock with how the PD handles the KEYSET command, and its that it discards the secure channel by default and sends an ACK on KEYSET in plain text.

In pd_decode_command function:

```
memcpy(pd->sc.scbk, cmd.keyset.data, 16);  
CLEAR_FLAG(pd, PD_FLAG_SC_USE_SCBKD);
CLEAR_FLAG(pd, OSDP_FLAG_INSTALL_MODE);  
sc_deactivate(pd); <---- This 
pd->reply_id = REPLY_ACK;  
ret = OSDP_PD_ERR_NONE;  
break;  
```

Looking at the git commit log, discarding the secure channel on KEYSET was implemented in the following commit:
_commit 923c9603 2021-07-14 22:37_
_Discard Secure Channel if a KEYSET is ACKed in plaintext_

> HID readers are responding to a KEYSET command with an ACK in plaintext instead of using the current session keys to acknowledge this command (which is the reasonable thing to do as the command itself was received encrypted with the old key). Since the spec doesn't say anything about this, both methods are technically correct.  
> 
> Make changes to CP so it allows ACKs in plaintext for KEYSET command in particular and make the PD implementation do this too as it makes the code look cleaner there (perhaps HID did it for the same reasons :D).  

Our problem here is the opposite. The current library does not allow ACK in plaintext on keyset at all, and upon receiving the ACK in plaintext, it will discard the ACK reply and reset the secure channel. Because the CP does not acknowledge the ACK, it does not register a successful KEYSET command and will not save the key it previously sent. Instead, it will generate new session keys based on the old key which is SCBK-D, but the PD now uses the the new key causing a missmatch, causing a loop of endless CHLNG messages. 

Now, my reading of the osdp standard,  and as noted in the commit log for this change, the correct way would be that the keyset ack should be sent in secure channel as well, as the specification says keyset can only be used in secure channel to begin with..alas this seems to be another  one of those cases where the specification lacks some clarity and theres multiple implementations out and about on how to handle this

**Solution**
The simplest solution as i see it would be to add an optional flag that can be set to disable the sc_deactivate when doing a KEYSET, and leaving it up to the user to decide how to handle it. Looking at how the CP side of libosdp handles it, my understanding is that it can handle KEYSET ack in both plaintext and secure channel.

sidcha, 2024-08-26:
Thanks for the excellent bug report and that's a fair ask. I was basically cutting a corner when working on the CP side of this issue and now it's back to byte me :).

Will take a look at this soonish.

sidcha, 2024-09-04:
@ldholmgren please check if [485e34d](https://github.com/goToMain/libosdp/commit/485e34d5500665d5fb64a77484df594bc5a2e4d1) fixes your problem.

closed 2024-09-08 by sidcha

ldholmgren, 2024-09-09:
Great, ill test it asap