osdp-dev/libosdp #195: Reconnecting the reader With-SC

boobika, 2024-09-13:
Hi,

I have an issue with reconnecting the reader.

**Describe the bug**
The reader is securely connected (With-SC), I disconnect it from the cable for a moment and then reconnect it. Unfortunately, the connection does not reestablish.

**Expected behavior**
Reconnecting the reader With-SC

**Testing code:**
```python
import time
import serial
from osdp import Channel, PDInfo, ControlPanel, LogLevel

class SerialChannel(Channel):
    def __init__(self, device: str, speed: int):
        self.dev = serial.Serial(device, speed, timeout=0.05)
    def read(self, max_bytes: int):
        return self.dev.read(max_bytes)
    def write(self, data: bytes):
        return self.dev.write(data)
    def flush(self):
        self.dev.flush()
    def __del__(self):
        self.dev.close()

def _create_pd_info_list(channel: Channel) -> list[PDInfo]:
    scbk = bytes.fromhex('A3 2B 7F C8 1E 22 D5 14 F1 67 4A 3C BD E2 9F 53')
    return [
        PDInfo(1, channel, scbk=scbk)
    ]

def app_run():
    channel = SerialChannel('/dev/ttyS0', 19200)
    pd_info = _create_pd_info_list(channel)
    cp = ControlPanel(pd_info, log_level=LogLevel.Debug)
    cp.start()
    cp.sc_wait_all()
    while True:
        for index, item in enumerate(pd_info):
            cp_event = cp.get_event(item.address, 0)
            if cp_event is not None:
                print(f"cp.get_event() = PD-{item.address}, index: {index}, event: {cp_event}")
                if cp_event['event'] == 1: # reader
                    print("RFID card detected")
            else:
                time.sleep(0.01)           

if __name__ == "__main__":
    app_run()
```

**Console output:**
[image]

sidcha, 2024-09-13:
The PD was marked offline because it didn’t respond for 8 seconds. After this period the CP will try to restart communication after 300 seconds.

So you should either wait 5 minutes for the connection to restart or restart the CP so they can start communicating right away.

boobika, 2024-09-13:
The communication does not resume after 5 minutes or even after additional time...

Question: By restarting the CP, do you mean terminating and reinitializing the CP?

sidcha, 2024-10-17:
I can see that the CP does come back out of offline state and tries to restart communication with the PD after 5 minutes. See:
```
OSDP: CP: PD-101: 2024-10-17T16:02:37Z   osdp_cp.c:897  [WARN ] No response in 200ms; probing (8)
OSDP: CP: PD-101: 2024-10-17T16:02:38Z   osdp_cp.c:901  [ERROR] Response timeout for CMD: POLL(60)
OSDP: CP: PD-101: 2024-10-17T16:02:38Z   osdp_cp.c:1210 [ERROR] Going offline for 300 seconds; Was in 'Online' state
OSDP: CP: PD-101: 2024-10-17T16:07:38Z   osdp_cp.c:897  [WARN ] No response in 200ms; probing (1)
```

Are you sure you waited 5 minutes?

sidcha, 2024-10-23:
Closing as not-reproducible. Feel free to re-open if you have more data.

closed 2024-10-23 by sidcha