osdp-dev/libosdp #199: Packet scan skipped?

alexanderturner, 2024-10-23:
**Describe the bug**
Possible erroneous data flags packet scan skipped and CP hangs.

All I'm seeing at this point is 
recOSDP: CP: PD-1: [2024-10-22T22:42:49Z] [DEBUG] src/osdp_phy.c:521: Packet scan skipped:1 mark:0

Following this, all OSDP operations hang / don't occur. Is this known or any idea what could cause these. Occurs after weeks of perfect operation

sidcha, 2024-11-09:
@alexanderturner, this doesn't ring any bells immediately. One possible reason is invalid data in the serial stream; since you mention the device was working fine for a long time, that's unlikely.

Also, this cannot cause the CP to just hang. Did you see any issues before or after a while? Would be helpful if you can provide more info to debug this issue.

sidcha, 2025-01-10:
@alexanderturner Checking back on this issue. Are you still facing this problem? If so, could you please reproduce the issue with the most recent master and provide debug logs? Check the [docs](https://libosdp.sidcha.dev/libosdp/debugging) on doing this.

sidcha, 2025-01-22:
Closing due to lack of context. Feel free to re-open when you do have more data.