osdp-dev/libosdp #206: osdp_FMT might be parsed incorrectly

Sympatron, 2024-11-15:
**Describe the bug**
I am currently having a discussion with an RSFID reader manufacturer who claims that the "Character Count" specified in the osdp_FMT reply should be 2 more than the actual length, because it says "Number of characters, including START, END, CKSUM" in the spec.

I would disagree, but I wanted to know if you have any experience with other OSDP readers and how they handle this?

In my opinion the osdp_FMT reply is extremely underspecified. in B.4 it says readers can either send card data as an "array of bits" or and "array of BCD characters". Whatever that means... 🤦

sidcha, 2024-11-15:
Unfortunately no, this is the first I am hearing about this problem. Does the RSFID reader actually send those two bytes over OSDP? If not, I don't see how they can expect the CP to know about it.

Also, by that logic, shouldn't it be 3 more than the actual length? (START + END + CKSUM)

Sympatron, 2024-11-15:
That's what I am trying to tell them, but they are not very cooperative...

Do you have any idea what START, END and CKSUM are supposed to be? They are not mentioned anywhere.

I tried to look for any reference implementation, but not even SIA seems to know what they meant or care about it. Their [reference implementation](https://github.com/Security-Industry-Association/libosdp-conformance) does not support `osdp_FMT` and their [verification test cases](https://github.com/Security-Industry-Association/osdp-verified) don't contain tests either...

`OSDP.NET` also just skips it, too.

sidcha, 2024-11-21:
I have no idea what START, END and CKSUM means nor have I encountered them. Perhaps @IDmachines or @rsgmodelworks  knows something?

RSFID guys seems to be the only ones privy to this knowledge and you should press them to explain their decisions -- which they should as you've paid for their product.

rsgmodelworks, 2024-11-22:
osdp_FMT is not to be used, it was scheduled for removal years ago. There are no known implementations of this in the wild.  Any implementation using this would be considered non-interoperable.

Sympatron, 2024-12-03:
Should it be removed from LibOSDP then?

rsgmodelworks, 2024-12-03:
I think it should be removed.  Per the SIA 2.2.2 spec, which is now available (December 2024) osdp_FMT has been deprecated.  See section 7.11.  If you want to be gentle about it, you could disable it when the PD reports an OSDP version capability 4 (See section B.17, capability 16 value 4 means "SIA 2.2.2".)

sidcha, 2024-12-05:
Removed, thank you @Sympatron and @rsgmodelworks.

closed 2024-12-05 by sidcha