osdp-dev/libosdp #210: Handling of CMD_ID, CMD_CAP

AlexNCoder, 2024-12-26:
Handling of CMD_ID, CMD_CAP
------------------
Hello everyone! I found that CMD_ID handling on PD is insufficient, maybe (func pd_decode_command, case CMD_ID). It sets ret to OSDP_PD_ERR_GENERIC when osdp-message from the CP (physical device,  not libosdp) can be without data-bytes. By the way, real card reader (physical device) can handle such messages. Am I wrong?

Maybe we can change that condition from “if (len != CMD_ID_DATA_LEN)” to “if (len > CMD_ID_DATA_LEN)”
It's all the same about CMD_CAP.

I've created a [pull request][1]

[1]: https://github.com/goToMain/libosdp/pull/209

sidcha, 2025-01-09:
> [..] when osdp-message from the CP (physical device, not libosdp) can be without data-bytes.

What makes you think so? Can you cite the specification please.

> By the way, real card reader (physical device) can handle such messages.

Not sure what you mean by "real" card reader here.. I thought they were all real :)

AlexNCoder, 2025-01-21:
1. Unfortunately, I haven’t specification of the devices (access control system STR20-IP, https://products.rubezh.ru/products/kontroller_str20_ip-3459/#download, CardReader RELVEST LTD PR-P03.M). I only have the user manual, in Russian(. But I had sniffed a traffic between osdp-devices. I’ll attach a small dump later. Maybe that devices deviate from OSDP standart?

2. I work with an OSDP system of one card reader, one control system module and an computer-emulated osdp-device (by libosdp).
Real, I meant which is not emulated :)

sidcha, 2025-01-22:
OSDP specification, section 6.4 (osdp_ID) and 6.5 (osdp_CAP) both have one data byte set to zero which requests the PD to send the "Standard Response". This is probably a future extension possibility that SIA included so the CP can ask the PD for some non-"Standard Responses" if needed.

The way I see it, any CP not including this byte (and PDs that expect it to be empty) have to be considered non-compliant. Please report this to the manufacturer.

AlexNCoder, 2025-01-24:
I agree. Thank you!

closed 2025-01-24 by sidcha