osdp-dev/libosdp #214: CP mode: Sequence number is incremented when repeating (retrying) a command

probatom, 2025-01-10:
**Describe the bug**
I have a setup where our application using osdp in CP mode can started quicker than the PD hardware (a HID Signo 40 badge reader in this case) seems to boot up (this is within milliseconds after the reader is powered). The reader seems to have missed the first command sent to is by libosdp; there is no response. But it looks libosdp does not recover nicely:
```
L897 [WARN ] No response in 200ms; probing (1)
L489 [ERROR] Packet sequence mismatch (1/0)
L1210 [ERROR] Going offline for 300 seconds; Was in 'ID-Request' state
L1220 [DEBUG] StateChange: [ID-Request] -> [Offline] (SC-Inactive)
```
The CP sends a new ID-Request command with incremented sequence number (1), but the PD responds with sequence number 0, which indeed should be used according the standard when starting a new communication session (e.g. after boot). Although the standard is not very elaborate on how this should work in corner cases like this (retries).

**Expected behavior**
Second ID-Request attempt (retry) should use the same initial sequence number (0) as the first attempt.

**Observed behavior**
See above, CP reports "Packet sequence mismatch".

**Additional context**
Library version: 3.0.8

I've patched the libosdp code locally, by adding `pd->seq_number -= 1;` in `cp_phy_state_update`, right before `LOG_WRN("No response in 200ms; probing (%d)", pd->phy_retry_count);`. Then the scenario appears to work fine (with a retried command):
```
L900 [WARN ] No response in 200ms; probing (1)
L680 [DEBUG] CMD: ID(61) REPLY: PDID(45)
L1223 [DEBUG] StateChange: [ID-Request] -> [Cap-Detect] (SC-Inactive)
...etc...
```
If I trigger lost packets (manually disconnect RS485 wires) after the PD came online (active communication session with SC enabled) the retries using this modified code seem to work fine. A few "No response in 200ms" messages are logged, but the session remains intact.

I suspect that the same change would be an improvement for the other case that returns `OSDP_CP_ERR_CAN_YIELD`: the case of receiving an osdp_BUSY reply. Repeating the command the command with the same sequence number should allow the PD to complete the command it was already processing, while (if I understand the specs correctly) sending the command again with incremented sequence number should be interpreted by the PD a fresh request (so it should start over again). Both should work, but not restarting the action seems the nicer behavior to me. But I have no scenario for triggering osdp_BUSY responses.

Also, I'm not familiar enough with libosdp internals to know that if after a OSDP_CP_ERR_CAN_YIELD return, the next command going to the PD will always be a repeat of the previous command (or a full session reset). Because if something else can come in between, my proposed code change ("undoing" the sequence number increase in that location) will be incorrect.

probatom, 2025-06-02:
Thanks for your response.

> > [...] The CP sends a new ID-Request command with incremented sequence number (1), but the PD responds with sequence number 0, which indeed should be used according the standard when starting a new communication session
> 
> I agree this is an edge case but the CP is the owner of the communication and it decides which sequence number to use. And the choice of using 0 to start a new session is also for the CP to make.

In the described scenario, the PD probably never saw the first message, so it can be argued that the PD shouldn't have replied at all to the "retry", since from the PD perspective the sequence is out of order (it is still waiting to see a valid initial message with sequence number 0). In either case, the current retry behavior of libosdp in CP mode for the initial message does not work (it would only work in the scenario where the PD did receive the initial message, but the reply got lost).

> > I've patched the libosdp code locally, by adding pd->seq_number -= 1;
> 
> This would work except for the case `seq_number == 0`; maybe it's not that big a deal as that happens rarely. I will see if we can come up with a different approach for dealing with sequence numbers.. it has caused quite some pain over the years.

This code is, if I'm not mistaken, only reached after the sequence was incremented before, so seq_number > 0 at this point. Meaning that for the initial packet, it will correctly revert the 1 back into 0. However, when the had number wrapped from 3 back to 1, this code will also revert it back to 0, while it should in that case revert back to 3. I copied this "revert by decrementing" approach from some other cases in osdp_phy.c, which might have the same problem (it cannot know if a "1" should be reverted to a "0" or a "3"). Might be part of the mentioned pain.

It's probably more logical to increment right before starting with a new message rather than right after sending a message. At least it avoids the issue of reverting an increment that in hindsight was done prematurely. Although this approach probably also has its own nasty edge cases...

> > I suspect that the same change would be an improvement for the other case that returns OSDP_CP_ERR_CAN_YIELD: the case of receiving an osdp_BUSY reply.
> 
> Probably not. Once again the specification would have us speculate: it says the CP must increment the sequence number to indicate "ACK" of the pervious response. And sending a command with the same sequence number would send the "same" response without actual processing taking place -- indicating that this is a error recovery method for on wire corruptions (checksum mismatches). If the PD responded with BUSY, it hadn't yet finalized the actual response yet so an incremented retry is warranted in that sense too.

For an osdp_BUSY, the standard states: "the ACU shall continue to repeat the command in its original form
until the PD returns something other than osdp_BUSY". So also no increment of the sequence number. This also follows from elsewhere in the standard, where it states: "If, on the other hand, the ACU elects to abandon the command that received the BUSY reply, the PD shall recognize this condition (new sequence number) and shall process the new command." So incrementing the sequence number will probably work fine in most cases, but it means the PD is supposed to discard any work done so far and start over. Only if it concerns a request that the PD inherently cannot complete within the allowed response timeout, no progress will be made anymore if every time the sequence is incremented.

> > Also, I'm not familiar enough with libosdp internals to know that if after a OSDP_CP_ERR_CAN_YIELD return, the next command going to the PD will always be a repeat of the previous command (or a full session reset).
> 
> The `OSDP_CP_ERR_CAN_YIELD` return code signals the higher levels that this PD is in-between command-response pairs. So the CP can move on to the next PD in shared channels if needed. There is a chance that the PD responds after the CP had timed out and yielded control to another PD.

Thanks for the explanation, it is now clear to me what "yielding" means in libosdp context (and totally makes sense; I wasn't really considering the multiple PD scenario when I initially looked at this). And by the way, a well-behaved PD should never respond after the timeout (that's why the osdp_BUSY reply exists in the protocol, so the PD can signal it won't be able to produce the real reply in its allowed transmission timeslot). The library can choose try to be lenient towards not-so-well-behaved devices, or strictly sticking to the protocol (just restarting the session, if needed to recover) -- I don't know what libosdp's philosophy is in this regard (also: off-topic for this issue report).

sidcha, 2025-09-21:
This should be fixed in 7f6d3c358efc. I tested it to work with unit tests, please feel free to re-open if this issue is not fixed for your use case.