osdp-dev/libosdp #222: OSDP Check Header Fail if 0x53 Present in Checksum

ep-jinyee, 2025-03-06:
**Describe the bug**
The bug is observed while I was using 4 PDs running using libosdp connecting to a CP running using libosdp, one of the PD goes offline and online repeatedly and consistently throughout the polling. This occurs in almost every 2s when observing. A side note, I did not enable security channel.

**Expected behavior**
All PDs once established (PDID and PD CAP), the polling should be consistently replied by PD. The PD should not swinging between online and offline.

**Observed behavior**
One of the PD will swinging between online and offline, every 2 seconds, which makes the whole channel unusable.

**A Comprehensive Log file**
This is PD 0 and PD 2, nothing all the packet seems normal

[image]

When PD 1 and PD 2 are on the bus, you can see it's complaining received address 0xff

[image]

Here's a screenshot from logic analyzer showing my controller (CP) is polling and reply from PDs. Clearly there's a gap between the poll and reply, and the CP report the PD went offline.

[image]

**Additional Context**
My first guest is since OSDP protocol packet is multidrop, when a CP is polling and PD replied to the CP, the reply will reach every PD. The other PDs check the address and discard the reply. However, imagine the scenario when the reply's checksum has byte such as 0xff, 0x53, the other PD will think this is a SOM and start consuming the next packet, yet it's not.

This screenshot of my logic analyzer, shows the reply from PD 1 has 0x53 at the checksum, and PD 0 replied with 0x41 which means NACK, then PD 0 went offline.

[image]

ep-jinyee, 2025-03-06:
I suggest we can consider changing the method of checking packet header in this function.

https://github.com/goToMain/libosdp/blob/7a754547e2f1c292cdbdc7a453ff84f6293e578e/src/osdp_phy.c#L339

ep-jinyee, 2025-03-06:
I proposed a state machine approach for the function,  I know it's kind of old school and dummy, but it solve the problem temporary and let it run first. I'm still testing this code.

``` 
static int phy_check_header(struct osdp_pd *pd)
{
	int pkt_len, len, target_len;
	struct osdp_packet_header *pkt;
	uint8_t cur_byte = 0;
	uint8_t *buf = pd->packet_buf;

	// we need to define a state here instead of just checking som
	enum {
		STATE_WAIT_MARK_OR_SOM,
		STATE_MARK_RECV_WAIT_SOM,
		STATE_SOM_RECV_WAIT_ADDR,
		STATE_ADDR_RECV_WAIT_LEN_LSB,
		STATE_LEN_LSB_RECV_WAIT_LEN_MSB,
		STATE_LEN_MSB_RECV_WAIT_CTRL,
	} state;

	state = STATE_WAIT_MARK_OR_SOM;
	int header_byte_count = 0;

	/* Scan for packet start */
	while (pd->packet_buf_len == 0) {
		if (osdp_rb_pop(&pd->rx_rb, &cur_byte)) {
			return OSDP_ERR_PKT_NO_DATA;
		}

		switch (state) {
		case STATE_WAIT_MARK_OR_SOM:
			if (cur_byte == OSDP_PKT_MARK) {
				state = STATE_MARK_RECV_WAIT_SOM;
			} else if (cur_byte == OSDP_PKT_SOM) {
				state = STATE_SOM_RECV_WAIT_ADDR;
			} else {
				return OSDP_ERR_PKT_WAIT;
			}
			buf[header_byte_count++] = cur_byte;
			break;
		case STATE_MARK_RECV_WAIT_SOM:
			if (cur_byte == OSDP_PKT_SOM) {
				state = STATE_SOM_RECV_WAIT_ADDR;
				buf[header_byte_count++] = cur_byte;
			} else {
				/* validate packet header */
				LOG_ERR("Invalid SOM 0x%02x", cur_byte);
				return OSDP_ERR_PKT_FMT;
			}
			break;
		case STATE_SOM_RECV_WAIT_ADDR:
			if (is_pd_mode(pd)) {
				if (cur_byte & 0x80) {
					LOG_WRN("Ignoring packet with invalid PD_ADDR.MSB");
					LOG_WRN("PD address cannot be %d, return OSDP_ERR_PKT_FMT",
						cur_byte);
					return OSDP_ERR_PKT_SKIP;
				}
			}

			if (is_cp_mode(pd)) {
				if (!(cur_byte & 0x80)) {
					LOG_WRN("Ignoring packet with invalid PD_ADDR.MSB");
					LOG_WRN("Invalid reply address %d, return OSDP_ERR_PKT_FMT",
						cur_byte);
					return OSDP_ERR_PKT_SKIP;
				}
			}
			state = STATE_ADDR_RECV_WAIT_LEN_LSB;
			buf[header_byte_count++] = cur_byte;
			break;
		case STATE_ADDR_RECV_WAIT_LEN_LSB:
			buf[header_byte_count++] = cur_byte;
			state = STATE_LEN_LSB_RECV_WAIT_LEN_MSB;
			break;
		case STATE_LEN_LSB_RECV_WAIT_LEN_MSB:
			buf[header_byte_count++] = cur_byte;
			state = STATE_LEN_MSB_RECV_WAIT_CTRL;
			break;
		case STATE_LEN_MSB_RECV_WAIT_CTRL:
			buf[header_byte_count++] = cur_byte;
			pd->packet_buf_len = header_byte_count;
			break;
		}
	}

	if (header_byte_count == sizeof(struct osdp_packet_header) + 1) {
		SET_FLAG(pd, PD_FLAG_PKT_HAS_MARK);
	} else {
		CLEAR_FLAG(pd, PD_FLAG_PKT_HAS_MARK);
	}

	pkt = (struct osdp_packet_header *)(buf + packet_has_mark(pd));

	/* Found start of a new packet; wait until we have atleast the header */
	target_len = sizeof(struct osdp_packet_header);
	pd->packet_buf_len += osdp_rb_pop_buf(&pd->rx_rb, buf + pd->packet_buf_len, target_len - pd->packet_buf_len);
	if (pd->packet_buf_len < target_len) {
		return OSDP_ERR_PKT_WAIT;
	}
	/* validate packet length */
	pkt_len = (pkt->len_msb << 8) | pkt->len_lsb;

	if (pkt_len > OSDP_PACKET_BUF_SIZE ||
	    (unsigned long)pkt_len < sizeof(struct osdp_packet_header) + 1) {
		return OSDP_ERR_PKT_FMT;
	}

	return pkt_len + packet_has_mark(pd);
}
```

sidcha, 2025-03-11:
@ep-jinyee, thanks for the code suggestion. But I think we can achieve what you intend with a little less invasive change. Please apply the following patch and see if it fixes your issue.

[0001-WIP-Fix-packet-scan-byte-skips.patch](https://github.com/user-attachments/files/19195751/0001-WIP-Fix-packet-scan-byte-skips.patch)

PS. I just eyeballed that code so please be prepared to fix any issues you might encounter. To properly validate this changes, I need to write a fuzzer but I don't have the time for it now.

commit dd7b12f9 (2025-03-28): Phy: Fix packet header scan

When looking for the start of a packet, we look for 0x53 in the byte
stream and then look for the rest of the bytes as a valid OSDP packet.
When the packet does not make sense, we just drop what we collected from
the channel thus far (one header width) and continue waiting for another
header.

This is problematic if the 0x53 happens to be in the last few bytes of a
packet which means we would have read the beginning on the next packet
and then dropped that too causing a potentially valid packet to be
dropped. To fix this, let's scan the bytes in cache for another 0 [... cut]
closed 2025-03-28 by sidcha