osdp-dev/libosdp #234: CP Python via serial not working

ac-tinker, 2025-05-30:
After doing a new install of the libosdp python package (v3.0.8 master, with python 3.11.2) communication can not be establish with a Mercury Security controller. My previous libosdp python package (v2.2.0 with python 3.9.3) was working without any major issues (sometimes some garbage messages)

**Expected behavior**
Expected behaviour is that my python script can connect to the Mercury Security EP1502 panel as a PD device and be able to receive commands and send events.

**Observed behavior**
With the new package the communication fails to setup a connection, some initials packets are sent/received but actual connection is not established.

OSDP: PD-0: 2025-05-30T21:23:20Z   osdp_pd.c:1142 [WARN ] SCBK not provided. PD is in INSTALL_MODE
OSDP: PD-0: 2025-05-30T21:23:20Z osdp_diag.c:34   [WARN ] Capturing packets to 'osdp-trace-pd-0-2025-05-30T21_23_20Z.pcap'
OSDP: PD-0: 2025-05-30T21:23:20Z osdp_diag.c:35   [WARN ] A graceful teardown of libosdp ctx is required for a complete trace file to be produced.
pyosdp: 2025-05-30T21:23:20Z   osdp_pd.c:1160 [INFO ] PD Setup complete; LibOSDP-3.0.8 master (8c334ad)
OSDP: PD-0: 2025-05-30T21:23:20Z  osdp_phy.c:73   [EMERG] RX ring buffer overflow!
OSDP: PD-0: 2025-05-30T21:23:20Z  osdp_phy.c:521  [DEBUG] Packet scan skipped:6 mark:1
OSDP: PD-0: 2025-05-30T21:23:21Z   osdp_pd.c:665  [DEBUG] CMD: ID(61) REPLY: PDID(45)
OSDP: PD-0: 2025-05-30T21:23:21Z  osdp_phy.c:73   [EMERG] RX ring buffer overflow!
OSDP: PD-0: 2025-05-30T21:23:21Z   osdp_pd.c:665  [DEBUG] CMD: ID(61) REPLY: PDID(45)
OSDP: PD-0: 2025-05-30T21:23:21Z  osdp_phy.c:73   [EMERG] RX ring buffer overflow!
OSDP: PD-0: 2025-05-30T21:23:21Z   osdp_pd.c:665  [DEBUG] CMD: ID(61) REPLY: PDID(45)
OSDP: PD-0: 2025-05-30T21:23:21Z  osdp_phy.c:73   [EMERG] RX ring buffer overflow!
OSDP: PD-0: 2025-05-30T21:23:21Z   osdp_pd.c:665  [DEBUG] CMD: ID(61) REPLY: PDID(45)
OSDP: PD-0: 2025-05-30T21:23:21Z  osdp_phy.c:73   [EMERG] RX ring buffer overflow!
OSDP: PD-0: 2025-05-30T21:23:21Z   osdp_pd.c:665  [DEBUG] CMD: ID(61) REPLY: PDID(45)
OSDP: PD-0: 2025-05-30T21:23:21Z  osdp_phy.c:73   [EMERG] RX ring buffer overflow!
OSDP: PD-0: 2025-05-30T21:23:21Z   osdp_pd.c:665  [DEBUG] CMD: ID(61) REPLY: PDID(45)
OSDP: PD-0: 2025-05-30T21:23:21Z   osdp_pd.c:665  [DEBUG] CMD: ID(61) REPLY: PDID(45)
OSDP: PD-0: 2025-05-30T21:23:21Z   osdp_pd.c:665  [DEBUG] CMD: ID(61) REPLY: PDID(45)
OSDP: PD-0: 2025-05-30T21:23:21Z   osdp_pd.c:665  [DEBUG] CMD: ID(61) REPLY: PDID(45)
OSDP: PD-0: 2025-05-30T21:23:21Z   osdp_pd.c:665  [DEBUG] CMD: ID(61) REPLY: PDID(45)
OSDP: PD-0: 2025-05-30T21:23:21Z  osdp_phy.c:423  [ERROR] Invalid checksum 00/ff
OSDP: PD-0: 2025-05-30T21:23:21Z   osdp_pd.c:1031 [ERROR] CMD receive error/timeout - err:-2
OSDP: PD-0: 2025-05-30T21:23:21Z   osdp_pd.c:665  [DEBUG] CMD: ID(61) REPLY: PDID(45)
OSDP: PD-0: 2025-05-30T21:23:21Z   osdp_pd.c:665  [DEBUG] CMD: ID(61) REPLY: PDID(45)
OSDP: PD-0: 2025-05-30T21:23:21Z   osdp_pd.c:1031 [ERROR] CMD receive error/timeout - err:-2
OSDP: PD-0: 2025-05-30T21:23:21Z  osdp_phy.c:521  [DEBUG] Packet scan skipped:6 mark:1
OSDP: PD-0: 2025-05-30T21:23:21Z  osdp_phy.c:423  [ERROR] Invalid checksum f1/00
OSDP: PD-0: 2025-05-30T21:23:21Z   osdp_pd.c:1031 [ERROR] CMD receive error/timeout - err:-2
OSDP: PD-0: 2025-05-30T21:23:21Z  osdp_phy.c:521  [DEBUG] Packet scan skipped:8 mark:1
OSDP: PD-0: 2025-05-30T21:23:21Z  osdp_phy.c:423  [ERROR] Invalid checksum f1/00
OSDP: PD-0: 2025-05-30T21:23:21Z   osdp_pd.c:1031 [ERROR] CMD receive error/timeout - err:-2
OSDP: PD-0: 2025-05-30T21:23:21Z  osdp_phy.c:521  [DEBUG] Packet scan skipped:10 mark:1
OSDP: PD-0: 2025-05-30T21:23:21Z  osdp_phy.c:415  [ERROR] Invalid crc 0x1786/0x0053
OSDP: PD-0: 2025-05-30T21:23:21Z   osdp_pd.c:1031 [ERROR] CMD receive error/timeout - err:-2
OSDP: PD-0: 2025-05-30T21:23:21Z   osdp_pd.c:1031 [ERROR] CMD receive error/timeout - err:-2
OSDP: PD-0: 2025-05-30T21:23:21Z  osdp_phy.c:521  [DEBUG] Packet scan skipped:12 mark:1
OSDP: PD-0: 2025-05-30T21:23:21Z  osdp_phy.c:423  [ERROR] Invalid checksum f1/00
OSDP: PD-0: 2025-05-30T21:23:21Z   osdp_pd.c:1031 [ERROR] CMD receive error/timeout - err:-2

**Additional context**
Communication is without secure channel, only 1 reader connected to reader port. However connecting the same panel to the older libosdp version works with no major issues.

osdp: PD: WARN : chn-0: SCBK not provided. PD is in INSTALL_MODE
osdp: PD: INFO : chn-0: Setup complete - libosdp-2.2.0 master (883067c)
pyosdp Version: libosdp-2.2.0 Info: master (883067c)
osdp: PD: DEBUG: chn-0: CMD: ID(61) REPLY: PDID(45)
osdp: PD: DEBUG: chn-0: CMD: CAP(62) REPLY: PDCAP(46)
PD received command:  {'command': 2, 'temporary': False, 'led_number': 0, 'reader': 0, 'control_code': 1, 'off_color': 0, 'on_color': 1, 'on_count': 10, 'off_count': 10}
osdp: PD: DEBUG: chn-0: CMD: LED(69) REPLY: ACK(40)
PD received command:  {'command': 4, 'control_code': 2, 'temp_time': 0, 'offset_col': 1, 'offset_row': 1, 'reader': 0, 'data': 'Ready               16:32       '}
osdp: PD: DEBUG: chn-0: CMD: TEXT(6b) REPLY: ACK(40)
osdp: PD: DEBUG: chn-0: CMD: LSTAT(64) REPLY: LSTATR(48)
osdp: PD: DEBUG: chn-0: CMD: RSTAT(67) REPLY: RSTATR(4b)
PD received command:  {'command': 4, 'control_code': 2, 'temp_time': 0, 'offset_col': 1, 'offset_row': 1, 'reader': 0, 'data': 'Ready               16:33       '}
osdp: PD: DEBUG: chn-0: CMD: TEXT(6b) REPLY: ACK(40)
PD received command:  {'command': 4, 'control_code': 2, 'temp_time': 0, 'offset_col': 1, 'offset_row': 1, 'reader': 0, 'data': 'Ready               16:34       '}
osdp: PD: DEBUG: chn-0: CMD: TEXT(6b) REPLY: ACK(40)
PD received command:  {'command': 4, 'control_code': 2, 'temp_time': 0, 'offset_col': 1, 'offset_row': 1, 'reader': 0, 'data': 'Ready               16:35       '}
osdp: PD: DEBUG: chn-0: CMD: TEXT(6b) REPLY: ACK(40)
PD received command:  {'command': 4, 'control_code': 2, 'temp_time': 0, 'offset_col': 1, 'offset_row': 1, 'reader': 0, 'data': 'Ready               16:36       '}
osdp: PD: DEBUG: chn-0: CMD: TEXT(6b) REPLY: ACK(40)

Hope you can help me out,

Bjorn

sidcha, 2025-06-02:
That is true, sorry I forgot that python API calls refresh internally. Can you try adding logs after `pd->channel.recv()` and after `osdp_rb_pop_buf()` calls to see they flow of packet data after it reached LibOSDP. It idea is to see where the hold up is -- the pop method should keep draining the ring buffer to make room for incoming data from channel.recv().

ac-tinker, 2025-06-02:
Sorry I'm not into c programming so I don't know how to put proper logging into these functions as they seem to work with structs and pointers.

sidcha, 2025-09-08:
@ac-tinker, this definitely looks like a regression and sorry about the delayed response, I was busy with another project.

If you have some time I can guide you through the debugging process. Basically, you'd have to compile the python module from sources after adding some printf-s at some places.

To install libosdp from source:
```bash
python3 -m venv .venv
source ./.venv/bin/activate
pip install --upgrade pip
pip install "./python"
```

Then, from that shell, to clean up:
```
deactivate
rm -rf .venv
```

You can repeat the above steps (install and clean up) between multiple printf() additions. If you are open to it, I can tell you where exactly you should place those printf() calls.

Also, please post your python program for verification.

sidcha, 2025-09-21:
@ac-tinker, I have pushed some general fixes and enhancements please check if that helps. Asides from that, I don't see a way to move forward on this without your help debugging. Closing as no-op

closed 2025-09-21 by sidcha