osdp-dev/libosdp #243: When PD continuously sends commands w/o interruption, the CP encounters `Packet sequence mismatch (3/0)`

kevinchan123456, 2025-09-16:
**Describe the bug**
I would like to inquire about an issue where when my PD (Peripheral Device) continuously sends commands without interruption, the CP (Control Panel) encounters this error. However, if the PD sends commands with a 1-second interval, there are no problems. I would like to ask how to resolve this issue.

**Expected behavior**
I want to continuously sends commands w/o interruption from PD.

**Additional context**
OSDP: CP: PD-1: 2025-09-16T07:35:02Z   osdp_cp.c:897  [WARN ] No response in 200ms; probing (1)
OSDP: CP: PD-1: 2025-09-16T07:35:03Z   osdp_cp.c:897  [WARN ] No response in 200ms; probing (2)
OSDP: CP: PD-1: 2025-09-16T07:35:04Z   osdp_cp.c:897  [WARN ] No response in 200ms; probing (3)
OSDP: CP: PD-1: 2025-09-16T07:35:05Z  osdp_phy.c:489  [ERROR] Packet sequence mismatch (3/0)
The error logs show that the OSDP (Open Supervised Device Protocol) communication is experiencing timeout issues and sequence number mismatches when commands are sent too rapidly without proper timing intervals.

[1]: https://libosdp.sidcha.dev/libosdp/debugging.html

kevinchan123456, 2025-09-19:
I wrote the minimum reproducible code for this issue.
**You need to run CP to send CMP_POLL, if you don't run cp, this bug will not happen.**

```
namespace debug {
	osdp_t *ctx = nullptr;
}

int OsdpSetup(const struct shell *shell)
{
    int flags = 0;
    uint8_t scbk[16] = {0};
    uint8_t *p_scbk = nullptr;
    int baud = 9600;
    int address = 1;

    shell_print(shell, "OSDP setup with baud=%d, address=%d\n", baud, address);

    if (syno_config::get_scbk(scbk))
    {
        shell_print(shell, "Using SCBK: %s\n", syno_utility::to_hex_string(scbk, 16).c_str());
        flags |= OSDP_FLAG_ENFORCE_SECURE; // Enable secure channel if SCBK is provided
        p_scbk = scbk;
    }
    else
    {
        shell_print(shell, "No SCBK found in config");
        flags |= OSDP_FLAG_INSTALL_MODE; // Install mode if no SCBK
        p_scbk = nullptr;
    }

    char sn[128] = {0};
    if (!anmw_get_serial_number(sn, sizeof(sn)))
    {
        anmw_log_error("Failed to get serial number\n");
        return -1;
    }

    osdp_pd_cap cap[] = {
        {OSDP_PD_CAP_CARD_DATA_FORMAT, 0x03, 0},
        {OSDP_PD_CAP_READER_LED_CONTROL, 0x05, 1},
        {OSDP_PD_CAP_READER_AUDIBLE_OUTPUT, 0x02, 1},
        {(uint8_t)-1, 0, 0},
    };

    osdp_pd_info_t info_pd = {
        .baud_rate = baud,
        .address = address,
        .flags = flags,
        .id = {
            .version = CONFIG_SYNO_READER_MODEL_VERSION,
            .model = CONFIG_SYNO_READER_MODEL_NUMBER,
            .vendor_code = 0xD00990,
            .serial_number = syno_utility::get_serial_number(),
            .firmware_version = syno_firmware::get_version_number_without_build(),
        },
        .cap = cap,
        .scbk = p_scbk,
    };

    debug::ctx = osdp_pd_setup(&info_pd);

    if (debug::ctx == nullptr)
    {
        anmw_log_error("osdp_pd_setup failed\n");
        return -1;
    }

    shell_print(shell, "pd setup completed\n");

    return 0;
}

static int cmd_send_osdp_event(const struct shell *shell, size_t argc, char **argv)
{
	if (debug::ctx != nullptr) osdp_pd_teardown(debug::ctx);

	int err = OsdpSetup(shell);
	if (err < 0) {
		shell_error(shell, "OSDP setup failed with error: %d\n", err);
		return -1;
	}

	//std::unique_ptr<osdp_event> event = std::make_unique<osdp_event>();
	osdp_event *event = (osdp_event *)k_malloc(sizeof(osdp_event));

	uint8_t payload_data[OSDP_EVENT_MFGREP_MAX_DATALEN];
	int data_len = 100;
	memset(payload_data, 'A', data_len);

	event->type = OSDP_EVENT_MFGREP;
	event->mfgrep.vendor_code = 0xD00990;
	event->mfgrep.command = 32;
	event->mfgrep.length = data_len;
	memcpy(event->mfgrep.data, payload_data, data_len);

	err = osdp_pd_notify_event(debug::ctx, event);
	err = osdp_pd_notify_event(debug::ctx, event);
	err = osdp_pd_notify_event(debug::ctx, event);

	shell_print(shell, "OSDP event sent with result: %d\n", err);

	return 0;
}
```

sidcha, 2025-09-19:
@kevinchan123456 Thanks for reporting the issue, can you please retry with the latest master and report your findings? I pushed a commit that fixes a bug in the code that that caused that panic.

sidcha, 2025-09-21:
7f6d3c358efc also should help with this issue. Closing for now, please feel free to re-open if the issue persists.

kevinchan123456, 2025-10-07:
After https://github.com/goToMain/libosdp/commit/7f6d3c358efc741f02de143782fde9e86d7abc35 merged, I still got this error.
But finally!!! I find the root cause.
```
## enable C++ support
#CONFIG_CPP=y
## expect to be C++23
#CONFIG_STD_CPP2B=y
## standard C++ library
#CONFIG_GLIBCXX_LIBCPP=y
```
The root cause is I enable the cpp config in my project.
After I change my programming language from cpp to c, this bus fault will not happend again.

Do you have any idea of this root cause?